adafruit/uf2-samdx1 #16: Metro M0 double-click difficult with v1.26.0-adafruit.0 and later

dhalbert, 2018-03-22:
Doing a double-click reset to get to UF2 (pulsing LED) on a Metro M0 with v1.23.1 is easy. But with 1.26.0-adafruit.0 and 2.0.0-adafruit.0, it has become much more difficult for me. I have to try multiple times. I verified by loading the older bootloader and it became easier again. I tried with various different user programs, including just a blinky.

I tried 2.0.0-adafruit.0 on Gemma M0, CPX, and Feather M0 Express, and did not have difficulty doing  double-click. I thought it might be `CRYSTALLESS` vs crystal-controlled clock, but the Feather M0 has a crystal and it's fine.

Speculation: Doing a double-click reset on the Metro M0 with the reset jumper still seems easy. So maybe it's some bounce characteristic of the larger switch vs a timing change since 1.23.1.

dhalbert, 2018-03-27:
@ladyada This double-click difficulty appears to be due to this code to make SWD programming more reliable:
https://github.com/adafruit/uf2-samd21/blob/c91ce29a4826f3e247a31bd193c0b2c8b6be9eea/src/main.c#L165-L168
```
#if (USB_VID == 0x239a) && (USB_PID == 0x0013)  // Adafruit Metro M0
    // Delay a bit so SWD programmer can have time to attach.
        delay(100);
#endif
```
This is right at the top of `main()`. If I comment out the `delay(100);`, double-click is much easier. How do you think we could satisfy both requirements? Maybe shorten the delay?

The Feather M0, etc. don't have this delay but seem to be programmable OK (I guess). Is there a difference between how those are factory-programmed and the Metro M0?

tannewt, 2018-03-27:
My guess is that this is due to the SWCLK line being shared with the
neopixel. Once the NeoPixel color is set the SWCLK line will be
disconnected internally.

A different fix would be to activate the gpio to output neopixel only when
new data needs to be sent and leaving it on mux G (SWCLK) otherwise.

tannewt, 2018-03-27:
We'd need to set the mux back at the end of this function: https://github.com/Microsoft/uf2-samd21/blob/1f706a115c3feb86d6f92c325f35d92981f6e54c/inc/neopixel.h#L63