adafruit/uf2-samdx1 #24: linker error

deanm1278, 2018-07-17:
I'm getting a linker error when trying to compile these bootloaders with arm-none-eabi-gcc 6.3.1:

> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMD21G18A__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/itsybitsy_m0 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/itsybitsy_m0 -Ilib/samd21/samd21a/include/ src/hid.c -o build/itsybitsy_m0/hid.o
arm-none-eabi-gcc -Lbuild/itsybitsy_m0 -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -Wall -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -save-temps -nostartfiles --specs=nano.specs --specs=nosys.specs \
	 -Tscripts/samd21j18a.ld \
	 -Wl,-Map,build/itsybitsy_m0/bootloader-itsybitsy_m0-v2.0.0-adafruit.5.map -o build/itsybitsy_m0/bootloader-itsybitsy_m0-v2.0.0-adafruit.5.elf build/itsybitsy_m0/flash_samd21.o build/itsybitsy_m0/init_samd21.o build/itsybitsy_m0/startup_samd21.o build/itsybitsy_m0/usart_sam_ba.o build/itsybitsy_m0/utils.o build/itsybitsy_m0/cdc_enumerate.o build/itsybitsy_m0/fat.o build/itsybitsy_m0/main.o build/itsybitsy_m0/msc.o build/itsybitsy_m0/sam_ba_monitor.o build/itsybitsy_m0/uart_driver.o build/itsybitsy_m0/hid.o
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: section .binfo LMA [0000000000001ff0,0000000000001fff] overlaps section .relocate LMA [0000000000001e50,000000000000201b]
collect2: error: ld returned 1 exit status
Makefile:117: recipe for target 'build/itsybitsy_m0/bootloader-itsybitsy_m0-v2.0.0-adafruit.5.bin' failed
make: *** [build/itsybitsy_m0/bootloader-itsybitsy_m0-v2.0.0-adafruit.5.bin] Error 1

dhalbert, 2018-07-17:
Yeah, you need to use gcc 7.x: https://github.com/Microsoft/uf2-samdx1/issues/50.

deanm1278, 2018-07-17:
@dhalbert thanks, didn't see that issue. I switched back to 4.8 :/

closed 2018-07-17 by deanm1278