adafruit/uf2-samdx1 #27: Error using update bootloader sketch for SAMD51

WalterPrech, 2018-08-09:
I downloaded the sketch for updating the bootloader of Metro M4 (update-bootloader-metro_m4-v2.0.0-adafruit.5.ino from https://github.com/adafruit/uf2-samdx1/releases/tag/v2.0.0-adafruit.5).

I'm using the Arduino IDE 1.8.5 on Mac OS X. 

If I open the sketch an click "Verify/Compile" many error messages occurs:
"update-bootloader-metro_m4-v2.0.0-adafruit.5:1062: error: 'volatile struct NVMCTRL_INTFLAG_Type::<anonymous>' has no member named 'READY'
     while (NVMCTRL->INTFLAG.bit.READY == 0) {...."

The sketch seems to be generated for the SAMD21 chip.

On the SAMD21 and SAMD51 the NVMCTRL_INTFLAG Types (and some others) are different.

If I download the uf2-samdx1 bootloader and compile the "update-bootloader-metro_m4-xxx.ino" the same error occurs.

ladyada, 2018-08-09:
oh we probably havent gotten that workin' yet - on the '51, you can use the updating UF2 for now

WalterPrech, 2018-08-09:
Thank you. Where can I download the files - I need the .ino file?

Do you change the uf2 bootloader sources, to customize the bootloader and on my side?

ladyada, 2018-08-09:
we dont have the ino file, it doesnt work. all files here are
https://github.com/adafruit/uf2-samdx1/releases
you can change uf2 bootloader all you like, just fork this repo and compile it yourself :)

closed 2018-08-09 by ladyada