adafruit/uf2-samdx1 #29: How to restore the bootloader on SAMD51?

tuupola, 2018-08-17:
It seems I accidentally broke the bootloader in Feather M4 while trying to flash an example code downloaded from Atmel Start.  At the moment I only get fast blinking yellow led. No USB device. Double tapping reset button does not bring bootloader mode. 

```
$ bossac --info --debug --port /dev/cu.usbmodem1411 --erase --write --verify --reset  AtmelStart.bin
Set binary mode
version()=v1.1 [Arduino:XYZ] Jun 25 2018 17:55:55
Connected at 921600 baud
readWord(addr=0)=0x2000d6a0
readWord(addr=0xe000ed00)=0x410fc241
readWord(addr=0x4)=0x58d
readWord(addr=0x41002018)=0x60060005
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x80)
writeWord(addr=0x20004020,value=0x20008000)
version()=v1.1 [Arduino:XYZ] Jun 25 2018 17:55:55
readByte(addr=0x804000)=0x39
readByte(addr=0x804001)=0x92
readByte(addr=0x804008)=0xff
readByte(addr=0x804009)=0xff
readByte(addr=0x80400a)=0xff
readByte(addr=0x80400b)=0xff
Device       : ATSAMD51x19
Version      : v1.1 [Arduino:XYZ] Jun 25 2018 17:55:55
Address      : 0x0
Pages        : 1024
Page Size    : 512 bytes
Total Size   : 512KB
Planes       : 1
Lock Regions : 32
Locked       : none
Security     : false
BOD          : false
BOR          : true
Erase flash
chipErase(addr=0)

Done in 5.005 seconds
Write 10656 bytes to flash (21 pages)
[                              ] 0% (0/21 pages)write(addr=0x20004034,size=0x1000)

SAM-BA operation failed
```

I reckon I could purchase a SWG programmer such as Black Magic Probe to reflash the bootloader. Are there any other options?

tannewt, 2018-08-18:
I'm a bit surprised its hosed since we usually set the bootloader protection bits.

It is a good thing to have a jlink when doing lower level work. The black magic probe didn't have samd51 support last I heard so make sure to check. The [JLink EDU mini](https://www.adafruit.com/product/3571) is inexpensive for non-commercial work. I use a [JLink BASE](https://www.adafruit.com/product/2209)

dhalbert, 2018-09-17:
Closing; assuming you've figured this out. Please reopen if not.

closed 2018-09-17 by dhalbert