adafruit/uf2-samdx1 #37: bin2uf2.js and uf2conv.py script offset issues?

FHeilmann, 2018-12-19:
Just searching through the code and noticed that both scripts mentioned assume the appstart at 0x2000. However, (unless I'm confused) when compiling for samd51, the code will end up being 0x4000 (16k) large. Shouldn't appstart be 0x4000 in that case?

tannewt, 2019-01-03:
Ya, it should be 0x4000 for samd51. In CircuitPython we use the py and provide the offset on the command line IIRC.

FHeilmann, 2019-01-08:
@ladyada @dhalbert what is your opinion on this? These false offsets bear the risk of people nuking their bootloaders on M4 boards.

dhalbert, 2019-01-09:
As @tannewt notes, the `Makefile` explicitly specifies the offset argument for `uf2conv.py` with `-b ~$(BOOTLOADER_SIZE). So unless you run this by hand, or use `binuf2.js`, it will not produce a UF2 with a bad offset.

We can, however, take away the default values for these arguments. `bin2uf2.js` doesn't take arguments but we could add those (or just get rid of this script in favor of the python one).

Also I'd note that the Makefile tries to use `python2` to run `uf2conv.py`. It should be made to be version independent or py3 only. @TG-Techie noticed that this morning.

FHeilmann, 2019-01-09:
I see, thanks for clearing that up for me. Removing the default values from these scripts would make them more foolproof in my opinion since you would not be able to execute them without specifically specifying an offset. In addition to your notes I also want to mention `sketch.cpp` again where `bootloader_size` is apssed as argument, but `BOOTLOADER_K` is hardcoded to be 8k

closed 2021-12-06 by FHeilmann