adafruit/uf2-samdx1 #43: SAME51J20A vs SAMD51J20A, any changed needed?

Timvrakas, 2019-02-01:
I'm bringing up a board I made that has a SAME51J20A. When I picked the SKU, I just assumed that the SAME51 components were implicitly in the "SAMD51" family. Based on the datasheet, the only difference between the SKUs is the CAN peripheral. With that in mind I compiled a .bin for the SAMD51J20A and flashed it to my SAME. It seems to work fine. Is there anything I'm missing, or is this just a naming issue?

Thanks
Tim

dhalbert, 2019-02-01:
Yes, looking at the datasheet, it appears there are no other differences to worry about.

The base chip is probably the same, and they turn the features on and off for marketing or testing reasons. For instance, if the CAN doesn't work on some, they'll turn it off with fuses and make those the SAMD chips. But even if it works they may turn it off anyway and sell it as a SAMD.

ladyada, 2019-05-05:
closed since no updates in a while

closed 2019-05-05 by ladyada