adafruit/uf2-samdx1 #47: Trouble with a NeoPixel on a Trinket M0 clone

alexwhittemore, 2019-03-22:
I'm trying to make a direct port of this bootloader to a board is, that for all bootloader intents and purposes, a Trinket M0, with a crystal and a neopixel instead of a dotstar.

board_config.h:
```c++
#ifndef BOARD_CONFIG_H
#define BOARD_CONFIG_H

#define CRYSTALLESS    0

#define VENDOR_NAME "Adafruit Industries"
#define PRODUCT_NAME "Trinket M0"
#define VOLUME_LABEL "TRINKETBOOT"
#define INDEX_URL "http://adafru.it/3500"
#define BOARD_ID "SAMD21E18A-Trinket-v0"

#define USB_VID 0x239A
#define USB_PID 0x001E

#define LED_PIN PIN_PA14 //NOTE- LED MOVED.
//#define LED_TX_PIN PIN_PA27
//#define LED_RX_PIN PIN_PB03

#define BOARD_NEOPIXEL_PIN PIN_PA16 // NOTE - SAME SERCOM DIFFERENT PIN
#define BOARD_NEOPIXEL_COUNT 1

#endif
```

The issue I'm having is that, on main.c line 138:
```c++
#if defined(__SAMD21E18A__)
    RGBLED_set_color(COLOR_LEAVE);
#endif
```
this `RGBLED_set_color(COLOR_LEAVE);` ALWAYS sets my neopixel full-bright-white. Whether I manually override the value, regardless of what I overwrite it with, etc. The LED writes on the way into other modes, like main.c 199: `RGBLED_set_color(COLOR_START);` all work fine. In fact, if I add a line on 200 to write whatever color I like, that works too. It's only on 139 that the write seems to fail.

If I comment out the LED write on 139, of course it doesn't happen - the LED goes red to indicate bootloader, then doesn't get cleared when the user code launches.

Also noteworthy, I tried this experiment on main.c 198:
```c++
    // not enumerated yet
    RGBLED_set_color(COLOR_START);
    delay(5000);
    RGBLED_set_color(0x0000FF);
    delay(10000);
    RGBLED_set_color(0x090900);
    led_tick_step = 10;
```

What's odd is that the `delay()`s don't work right. I had to set them to 5 and 10 seconds just to be perceptible - both of those are still well under a second. Perhaps my clock delcaration up there is off somehow? I also tried simply commenting out `#define CRYSTALLESS    1` as is the scheme in the Metro M0 configuration, but that didn't change any behavior. The timing does APPEAR to be different if I have `#define CRYSTALLESS    1` enabled vs disabled, but I can't tell how since the timings are so short as-is.

And finally, just for fun, if I change line 138 to:
```c++
#if defined(__SAMD21E18A__)
    RGBLED_set_color(COLOR_LEAVE);
    delay(5000);
    RGBLED_set_color(0x0000FF);
    delay(10000);
    RGBLED_set_color(0x090900);
#endif
```
the LED still goes full bright white. One would logically assume it DOES get written 3 times, white each time, and I just can't tell that.

I DID just figure out a second ago that in that ludicrous last case, it takes the bootloader 27 seconds after a reset button hit to enter user code. So maybe there's something funky going on like the clocking gets reconfigured by the time it gets to 138, and therefore the LED write timing is all messed up?

alexwhittemore, 2019-03-23:
A LITTLE more info: that goofy case for line 138 (last code block above) causes the bootloader to take ~23s to hit user code if `#define CRYSTALLESS    1`, and about ~27s if `#define CRYSTALLESS    0`

dhalbert, 2019-03-23:
The bootloader is now always crystalless: see `inc/uf2.h`, which sets it to `1` if it's not defined. Also, `src/init_samd21.c` does an `#if defined(CRYSTALLESS)` check. So the value doesn't actually matter. Setting it to zero will not turn off `CRYSTALLESS`. That's kind of a  bug: that `#if` check should be `#if CRYSTALLESS`.

It does sound like your clocks are messed up in some way, running very slowly. That would explain the NeoPixel color oddities.

dhalbert, 2019-03-23:
Note the commented-out lines here: https://github.com/adafruit/uf2-samdx1/blob/master/src/init_samd21.c#L124, which will output GCLK0 to pin PA30 for debugging: you can check the clock speed with a logic analyzer or scope. And you don't have to use PA30: GLCK_IO[0] is available on other pins as well for the SAMD21E, as listed in Table 7-1 in the SAMx21 datasheet.

alexwhittemore, 2019-03-23:
Great info, thanks! I'll try that to see if it reveals any particularly interesting info. But assuming the clock IS messed up, why would it be messed up *at that point* and not during the entire bootloader operation?

You're super helpful, by the way, thanks!

alexwhittemore, 2019-03-23:
Hmmm... With those two lines uncommented, I get 0V DC out of SWCLK. After a manual reset with the debugger disconnected, so no funny business with the pin muxes being configured for SWD still, either. Is it possible for the clock to be that badly misconfigured, and the core is falling back to a different clock?

alexwhittemore, 2019-03-23:
Oh. How about this. The bootloader clock works just fine and dandy, IFF the SAMD is connected to a USB host. I just looked through init_samd21.c and saw some notes about "USB correction" and thought to try that. 

As it happens, I'm trying to bootload primarily via UART, ultimately. Thoughts on how to get the clock working right, WITHOUT the guarantee of USB sync?

alexwhittemore, 2019-04-16:
Bit of a status update: Originally, my issue was that the neopixel would get stuck on full-white until functional application code did something with it. My solution to this point has been "comment out the neopixel" in the bootloader, so it simply doesn't get used.

My problem now is that the same clock problem is fouling up UART loading, which I also need.

The clocking problem isn't present when connected to native USB, as mentioned above, because the USB hardware conditions the main clock PLL with its 1ms sync pulse - otherwise, even with a good system clock, usb would work poorly. This isn't doing me much good, since bootloading over UART of course implies I don't want any USB connected.

Simply uncommenting the two lines @dhalbert mentioned didn't do the trick, because as line 117 above them states,

` // Add GCLK_GENCTRL_OE below to output GCLK0 on the SWCLK pin.`

In other words, line 117+:
// Add GCLK_GENCTRL_OE below to output GCLK0 on the SWCLK pin.
```c++
GCLK->GENCTRL.reg =
        GCLK_GENCTRL_ID(0) | GCLK_GENCTRL_SRC_DFLL48M | GCLK_GENCTRL_IDC | GCLK_GENCTRL_GENEN;
gclk_sync();
```
should become
```c++
GCLK->GENCTRL.reg =
        GCLK_GENCTRL_ID(0) | GCLK_GENCTRL_SRC_DFLL48M | GCLK_GENCTRL_IDC | GCLK_GENCTRL_GENEN | GCLK_GENCTRL_OE;
gclk_sync();
```
and THEN the clock output will work with the other lines (125+127) uncommented.

Ok, cool. I took that a step further and added that code to a generic led blink sketch, where it also works. So now I've got clock output both in (broken) bootloader and (functional) sketch.

What I've found with this new output is that, with USB plugged in, GCLK0 is quite precisely 48MHz in BOTH situations, and more like 47.15MHz with USB unplugged. And with USB unplugged, I can raise and lower that 47.15MHz by about .05MHz by breathing softly or blowing hard on the uC, so that's neat.

In other words, the 48MHz system clock works just fine and dandy, whether plugged in via USB or not, and in-sketch or not. So something else odd is going on.

I also noticed that the LED_BUILTIN breathing is much slower and smoother when plugged into USB, and kind of frantic (maybe 3Hz?) when not. That animation in the bootloader is driven by 

`void SysTick_Handler(void) { LED_TICK(); }`

so something is going wrong with whatever drives that interrupt. That's about where I'm at.

alexwhittemore, 2019-04-16:
(Sorry, this is as much for my own notes now as anything, but I'm sure someone at some point will find it interesting).

I just did another little experiment on my crystal-enabled board. I overrode the crystalless override on uf2.h:22 to allow crystal or not. 

Crystal enabled: ~47.97MHz, USB or not.
Crystal disabled: ~46MHz no USB, changing to 48.000MHz when USB plugged in.

Which strikes me as odd, because init_samd21.c:104 `SYSCTRL->DFLLCTRL.reg |= SYSCTRL_DFLLCTRL_ENABLE;` seems to me like it should enable USB syncing even if the crystal is used.

But in any event, regardless of the accuracy of the 48MHz clock, the systick handler seems to fire with the wrong frequency in any case.

alexwhittemore, 2019-04-17:
I added some lines at the beginning of of `led_tick()` which is the systick handler in the bootloader:

```c++
void led_tick() {
    LED_MSC_ON();
    LED_MSC_OFF();
```
so now I get a little runt pulse every time systick fires. It's showing me that, regardless of how fast the LED is breathing (slow when USB connected, frantic when independent), the systick handler is firing at 48kHz, best my cursor can tell. Which, of course, jives perfectly with `SysTick_Config(1000);` (48MHz/1000 = 48kHz).

So now I'm just more confused than when I began.

nekuneko, 2020-04-23:
I have the same Neopixel stuck issue in high white bright on a SAMD21E18A custom board on PA18 pin. But I think it may be caused beacuse of a bootloader compilation issue, because I uploaded the latest bootloader `bootloader-feather_m0_express-v3.9.0.bin` released here to my official _Adafruit Feather M0 Express_ and get the same problem on with the builtin neopixel brighting in high yellow. Nevertheless flashing the `bootloader-feather_m0-v2.0.0-adafruit.5.elf` bootloader this issue doesn't happens.

My solution was develop a patch for the ArduinoCore-samd in order to keep power off Neopixels at Arduino skechtes code startup. #223

dhalbert, 2020-05-17:
Should be fixed by #117. Please reopen if not true.

closed 2020-05-17 by dhalbert