adafruit/uf2-samdx1 #53: GrandCentral MacOS 10.14.4 upload problem

DaveCalaway, 2019-04-19:
Hi everyone, 
i have a problem with the GrandCentral and my MacOS 10.14.4. 
I seen this problem in the git https://tinyurl.com/y43gsxwu , but when i update the uf2 to 3.3.0 , the board resent itself during the download of the code in the Arduino IDE.
https://ibb.co/GcsmxK7

ladyada, 2019-04-19:
thats not unusual, just select the board from the Com port menu after upload.

DaveCalaway, 2019-04-19:
Those are my steps:
Put the board in the Bootloader mode, the NeoPixel become green.
I upload the code over arduino IDE, the IDE arrives at this point:

```
Sketch uses 12304 bytes (1%) of program storage space. Maximum is 1032192 bytes.
Device       : ATSAMD51x20
Version      : v1.1 [Arduino:XYZ] Apr 14 2019 22:38:13
Address      : 0x0
Pages        : 2048
Page Size    : 512 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 32
Locked       : none
Security     : false
BOD          : false
BOR          : true
Write 12560 bytes to flash (25 pages)

[                              ] 0% (0/25 pages)
[=========                     ] 32% (8/25 pages)
[===================           ] 64% (16/25 pages)
[============================  ] 96% (24/25 pages)
[==============================] 100% (25/25 pages)
Done in 0.176 seconds
Verify 12560 bytes of flash

[=                             ] 4% (1/25 pages)
[==                            ] 8% (2/25 pages)
[===                           ] 12% (3/25 pages)
[====                          ] 16% (4/25 pages)
[======                        ] 20% (5/25 pages)
[=======                       ] 24% (6/25 pages)
[========                      ] 28% (7/25 pages)
[=========                     ] 32% (8/25 pages)
[==========                    ] 36% (9/25 pages)
[============                  ] 40% (10/25 pages)
[=============                 ] 44% (11/25 pages)
[==============                ] 48% (12/25 pages)
[===============               ] 52% (13/25 pages)
[================              ] 56% (14/25 pages)
[==================            ] 60% (15/25 pages)
[===================           ] 64% (16/25 pages)
[====================          ] 68% (17/25 pages)
[=====================         ] 72% (18/25 pages)
[======================        ] 76% (19/25 pages)
[========================      ] 80% (20/25 pages)
[=========================     ] 84% (21/25 pages)
[==========================    ] 88% (22/25 pages)
[===========================   ] 92% (23/25 pages)
[============================  ] 96% (24/25 pages)
[==============================] 100% (25/25 pages)
Verify successful
Done in 0.318 seconds
```
Now the Mac give me that warning: 
[image]
and the NeoPixel turn off. The serial port disappears.

If i wait 5 seconds, the Arduino IDE give me another error:

`Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.`

Is that normal? When i need to change the port?

dhalbert, 2019-04-20:
What is listed in the Port Menu when GCM4BOOT is present, and what is listed when the Arduino program is running?

dhalbert, 2019-04-20:
Also, could you check your GC M4 to make sure the clock crystal is not upside down? There have been at least a couple soldered incorrectly. See pictures here:
https://forums.adafruit.com/viewtopic.php?f=63&t=150581

DaveCalaway, 2019-04-20:
I have tried with a Windows machine, i have the same problem.

I seen the post you linked to me, I have the same wrong component, the clock. 

How can i request another board?

dhalbert, 2019-04-20:
Start a thread in https://forums.adafruit.com/viewforum.php?f=63, with a picture of the upside-down component, and we can authorize a replacement there.

Sorry about this: apparently some crystals were upside down in the component tape reel used with the pick-n-place machine. The boards passed test, but don't work in many cases. I'll close this issue, since it's not software-related.

closed 2019-04-20 by dhalbert