adafruit/uf2-samdx1 #54: SERCOM clocks (20 and 19?) are wrong if 32kHz crystal is used

alexwhittemore, 2019-04-20:
Chip: SAMD21E18. Board: custom, looks like a Metro M0 or so, with crystal.
I've enabled hardware UART, but I'm having stability problems I think related to baud mismatch with a sensitive host. Might not actually be the cause, but let's pretend it is. If I try to force crystal mode (ext. 32.768kHz) to make the SAMD clock a little more accurate/stable, UART breaks entirely. If I add a little message to the bootloader

```c++
#if USE_UART
    /* UART is enabled in all cases */
    usart_open();
    usart_putdata("ready", 5); // My little tester
#endif
```

I can see that the SAMD sends "ready" at 78 baud, when it should be 115200. Note that it's close enough to 115200 when "crystalless". (78/115200)*48e6 is 32500, so my sneaking suspicion is that the clock configuration for crystalless vs crystal accidentally routes the low frequency input clock to general clock ID 20, which is what the SERCOM is clocked with, instead of the 48MHz system clock. I haven't yet confirmed or tracked down the code I suspect.

This is mostly a placeholder for me to fix myself later if I decide I need to, since I appear to be one of the 1% who want to use a crystal in bootloader, of the 2% who want to use hardware UART.

jpmeijers, 2019-10-27:
I'm actually running into this same issue. When using the default CRYSTALLESS configuration my UART runs at 122950 baud likely due to an inaccurate internal oscillator. I have a crystal on my board, so tried setting CRYSTALLESS to 0, which did not immediately help.

In `init_samd21.c` we see `#if defined(CRYSTALLESS)`, which will always match no matter the value of CRYSTALLESS. Because uf2.h does this:
```
// always go for crystalless - smaller and more compatible
#ifndef CRYSTALLESS
#define CRYSTALLESS 1
#endif
```

I changed init_samd21.c to be:
```
#if !defined CRYSTALLESS
  #error CRYSTALLESS is not defined.

#elif(CRYSTALLESS == 1)
```

so that the #else will be used when I have a crystal. 

Now I am running into the same issue as @alexwhittemore where my uart baudrate is orders of magnitudes incorrect. I have no idea where to start to fix this. Any help would be really appreciated.

dhalbert, 2019-10-28:
The clock setup is in https://github.com/adafruit/uf2-samdx1/blob/master/src/init_samd21.c. The non-`CRYSTALLESS` setup is part of the #ifdef-#else there, so I'm just guessing that there might be something wrong in that code. Note lines 124-127, which when uncommented will send GCLK0 to a pin, so you can monitor what its frequency actually is.

jpmeijers, 2019-10-28:
> so I'm just guessing that there might be something wrong in that code.

That's what I believe too. But I don't have the knowledge to debug this and therefore I'm looking for help.

> when uncommented will send GCLK0 to a pin,

Unfortunately 48MHz is a bit too high for any of my equipment to measure.

xyzzy42, 2022-09-15:
The baud rate divisor is just wrong, the base clock was changed without updating the divisor, this is fixed in unmerged PR #188 by 1f1c1883ae902ec8da3f9c406bbcb3c7f9892e8b

dhalbert, 2023-02-03:
Fixed by #188.

closed 2023-02-03 by dhalbert