adafruit/uf2-samdx1 #65: Changes needed for SAMD21E15

timmit99, 2019-05-29:
What would need to be done to make a version of this bootloader that works on a SAMD21E15? I have 2 chips on breakouts, both a SAMD21E15 and a ATSAMD21E18 the Trinket M0 bootloader works fine on the *E18 but not on the *E15.

I'm not sure what would have to be done to make this work. The only difference between the two chips is the capacities of the two:
 - SAMD21E18A has 256KB of Program memory and 32 KB of Ram
 - SAMD21E15B has 32KB of Program memory and 4 KB of Ram

I am not much of a programmer but i can take some guidance and see if i can make it work but i do not know where to start.

dhalbert, 2019-05-29:
4kB of RAM is not much. The bootloader code probably assumes a much larger available amount. The stack alone might use that up.

What is your application for this small chip? How does using the UF2 bootloader help this application, as opposed to using, say, the SAM-BA bootloader, or using no bootloader at all?

timmit99, 2019-05-30:
I am working on a work project and i have a bunch of the E15's from another retired work project. I was hoping to get this bootable so i can program it in arduino IDE/Atmel Studio over USB without having to be near the board to press the bootloader button/switch. 
The code itself is very lightweight ( 10kB) so having a smaller/cheeper chip in here would be best and not having to rebuy chips is preferable
I dont know what would be entailed but if i could disable the USB mass storage feature that would hopefully decrease the program size. At that point i really dont know what other options are out there for bootloaders.

dhalbert, 2019-05-30:
If you just want Arduino compatibility, you can make an Arduino bootloader using the sources here: https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero. That will probably be easier, though I don't know if it would run on such a small RAM chip. But consider whether it is really worth it, given your time and the cost of new chips/boards.

ladyada, 2019-06-30:
hiya gonna close this because its probably not going to work on the '15 and we have no plans to support that chip. please try the bossa bootloader above, it may work? or you may have to buy new chips

closed 2019-06-30 by ladyada