adafruit/uf2-samdx1 #67: Builds for SAMD51G19A broken - unknown cause

wallarug, 2019-06-18:
Afternoon,

I have been trying to build a bootloader for the SAMD51G19A but hit a small road block.

I forked the repository on Wednesday (last week).  My builds from that time have a version code of `v3.6.0.0-2-g4fdf0fe`.    They work perfectly fine on the board.

I then ran `git submodule update` about 2 hours ago and now all the builds I make do not work on the board.  They have version code `v3.6.0-4-g09c839b-dirty`.

I cannot find what changed after the sub-module update.  According to the operating system nothing changed except for the builds.  I can clearly see in the console history that this change happened after ran git submodule update.

According to the debugger, the bootloader is crashing on line 480 of src/cdc_enumerate.c.

There are no obvious changes to any submodules or this repository (except for one on June 14).  Is anyone able to explain how the version code gets generated and then I might be able to back trace this issue.

Thank you

tannewt, 2019-06-18:
What does `git status` show? The -dirty originates from `git describe`.

wallarug, 2019-06-19:
[image]

I don't think this is any help.  

FYI:  I included the build file.  This is not a mistake, just a temporary change.

dhalbert, 2019-06-19:
Could you copy/paste th text instead of a screen shot? I have color-vision issues and can't read the text in the image above.

I cloned your repo, did `make clean`, and built an `itsybitsy_m4` bootloader, which should be similar to yours: it's SAMD51G19A. It loaded fine and is running.

What version of arm gcc are you using? We're using 7.3.1. There were no submodule repo changes that I know of.

Could you clone your repo fresh, roll back a few commits, to say, 4fdf0fe, and see if it's still bad? Then advance to e88dd3a, and then 09c839b (with all the `.o` checkins, etc.). Maybe you did this already?

wallarug, 2019-06-21:
Hi @dhalbert  and @tannewt ,

I've done a bit of investigation and collaboration.  I think it is just the way the computer is configured.  I can build good builds but my colleague cannot.  Even on the same commit.

We will investigate further together.  Since we have a working build now.

closed 2019-06-21 by wallarug