adafruit/uf2-samdx1 #69: GCM4 Boot mode and L led

GMagician, 2019-06-18:
Grand central M4, when in boot mode, turn on (fading) the L led. This led is connected to D13 that may be an used pin, and none know what is connected to it...is not too dangerous to light it?
Isn't better too use the TX/RX yellow led that is connected only to a led?

ladyada, 2019-06-18:
blinking D13 is standard arduino behavior, no plans to change it (but feel free to make your own bootloader if you like!)

closed 2019-06-18 by ladyada

GMagician, 2019-06-19:
@ladyada please I think you have to reconsider what I wrote.
I know for sure that mega2560 and due have such led (called L) and for sure on due is connected in the same way. What I say is how such led is used. Are you really sure that during code upload such led is lighted?

ladyada, 2019-06-19:
it pulses during the bootloader to indicate that its in bootloader mode, it definitely works - we use it on all our boards!

GMagician, 2019-06-19:
of course it works but my question was a little bit different, but that's ok