adafruit/uf2-samdx1 #73: [Question] is 'setBootProt' dangerous?

GMagician, 2019-06-29:
Hello,

I'm creating code to program SEE size and I'm getting some hints from function above.

When fuses to set boot code protection are to be written, code does:
`exec_cmd(NVMCTRL_CTRLB_CMD_EP);`
this will format a page so 512 bytes set to 0xFF (but why format? if we conly clear bits no format is required)
program then write fuses[0] and fuses[1] (first 64 bits)
and finally:
`exec_cmd(NVMCTRL_CTRLB_CMD_WQW);`

**but SAMD51 datasheets writes:**

> 159:128 Reserved Factory settings - do not change.

so do we destroy something?

ladyada, 2019-06-29:
not sure, it could be wrong - if you have other code that works please submit it as a PR

GMagician, 2019-06-29:
Yes, I'm not sure as well, that's why I asked

GMagician, 2019-06-29:
@ladyada may you check in a brand new board, loading a boot code without fuse programming, what are the content in user page (512 bytes), I think my board has lost them, since fuse are programmed as soon as board starts

dhalbert, 2019-06-29:
The SAMD51 data sheet says:
>Note:  Before erasing the NVM User Page, ensure that the first 32 Bytes are read to a buffer and later
written back to the same area unless a configuration change is intended.

So I think you're right, `fuses` should be eight words (32 bytes) instead of just two, for the SAMD51.

The SAMD21 makes the calibration data unwritable, and it's in a different row.

The fuse fixing code was added because we were seeing broken boards where the entire user page was set to all 1's (by some unknown cause), which was enabling the watchdog timer. We only reset the first two words if they appear to have been completely smashed (all 1's).

You're right that a factory-fresh chip would only need BOOTPROT bits to be cleared, with no page erase necessary. But someone could also want to use an update-bootloader UF2 after using Atmel Studio, etc., and so BOOTPROT might not be pristine.

@ladyada I'm not sure what the factory programmers are doing in terms of preserving bytes past the first two words.

GMagician, 2019-06-29:
@dhalbert not only first 8 words (32 bits) but more, user page is 512 and when you format it you lose also some "personal" data stored by users.

> The remaining 480 Bytes can be used for storing custom parameters.

The problem is that if for some reason a board has executed the "restore" fuses code then we lost all "factory defaults" and I don't know what is the use of that.

GMagician, 2019-07-02:
Proposed a PR..I close this

closed 2019-07-02 by GMagician