adafruit/uf2-samdx1 #76: Changes to make SAMD51G18A work

SimonMerrett, 2019-07-24:
@dhalbert and anyone else - sorry this isn't an issue with your working code but I was hoping you could help. I have been developing with a Metro M4 express and then moved on to the ItsyBitsy M4 and now I have finally moved to my own custom board, I have realised I have the SAMD51G18A variant, rather than the 19A that only now do I see is in the dev boards that preceded it. Which parts of which files do I need to change to make a bootloader for the 18A variant? Help!

SimonMerrett, 2019-07-29:
I now think I have made a working bootloader for the SAMD51G18A. Is anyone interested in including it here (I will have to do a bit of learning on the PR / merge process)?

dhalbert, 2019-07-29:
If you think your board is going to be used by more than just you, then we're happy to entertain a PR. Otherwise it can just exist in your fork.

To do a PR, fork this repo, and create a branch in your own repo. Push the changes to that branch in your fork. Then you can create a pull request from the branch to master in this repo. More details here (in the context of CircuitPython, but it should be similar): https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

dhalbert, 2019-12-22:
Closing to await PR.

closed 2019-12-22 by dhalbert

mewa, 2020-05-15:
@SimonMerrett Do you mind sharing it? I'm developing one at the moment and struggle to make it flash from UF2. I get to boot to bootloader and see an MSC device, but when copying `update-bootloader-*.elf` it just stops working (i.e. is no longer enumerable) and I need to reflash it with Atmel-ICE.

@dhalbert Apart from the linker scripts do you see any obvious changes that need to be made? I have a draft version [here](https://github.com/adafruit/uf2-samdx1/compare/master...mewa:samd51x18), if you would kindly check it I would be really grateful.

dhalbert, 2020-05-15:
> @dhalbert Apart from the linker scripts do you see any obvious changes that need to be made? I have a draft version [here](https://github.com/adafruit/uf2-samdx1/compare/master...mewa:samd51x18), if you would kindly check it I would be really grateful.

Your draft looks OK, but I do want to hear that it works for you, and of course, you'll make new linker scripts.

I would prefer that you use `<something>k` constants in the linker file for sizes, instead of hex numbers, because they are much easier to read.

mewa, 2020-05-15:
Thank you for your feedback. Of course, I'll be happy to submit a cleaner PR once I have confidence it works. I just wanted to highlight the changes compared to the SAMD51x19 version

For now the MSC bootloader seems to work, at least when updating itself. Just to confirm, I'll also check how it performs with a different program, that stays in the user space. I will let you know once I have some answers.

SimonMerrett, 2020-05-15:
Hey @mewa I will gladly share. ATM the files are on a PC whose power supply is being used by another machine (as in cable, not ATX unit!). That machine is working on something right now but I will swap power over when its done and try to share. I never made a clone of the Adafruit repo, just downloaded, so I may just email you the files as a PR seems tricky ATM.