adafruit/uf2-samdx1 #79: Grand Central Issue - Klipper 3D Printer Firmware

jalanjarosz, 2019-07-29:
I'm attempting to use the Grand Central M4 as a replacement for the MEGA 2560/RAMPS combination.

Per instructions on the Klipper FW project, which supports the SAMD51 chipsets, I configured Klipper with the following selactable MCU options:

```
[*] Enable extra low-level configuration options
Micro-controller Architecture (SAMD21/SAMD51) --->
Processor model (SAMD51P20 (Adafruit Grand Central)) --->
Clock Reference (32.768Khz crystal) --->
Bootloader offset (16KiB bootloader (Arduino M0)) --->
USB ids --->
[ ] Specify a custom step pulse duration
() GPIO pins to set at micro-controller startup 
```

compiled the base FW and loaded it on the controller using bossa command:
`bossac -u -p COM7 --offset=0x4000 -w .\Metro\klipper.bin -v -b -R`

making of the klipper.bin  file was successful and the flash was also successful.

When getting the Raspberry Pi (main controller of Klipper) to talk to the MCU, I'm receiving the following in Klipper's log:

```
Starting serial connect
Loaded MCU 'mcu' 75 commands (v0.7.0-445-g2d86448-dirty-20190726_152807-octopiper / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'mcu' config: SOFT_PWM_MAX=256 STEP_DELAY=2 ADC_MAX=4095 RESERVE_PINS_crystal=PA0,PA1 CLOCK_FREQ=120000000 RESERVE_PINS_USB=PA24,PA25 MCU=samd51p20a STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
got {'#receive_time': 3521.247487406, u'crc': 2681911713L, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent_time': 3521.246409177, u'move_count': 1024}
Configured MCU 'STATS_SUMSQ_BASE' (1024 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Stats 3521.3: gcodein=5 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=3491 bytes_read=3613 bytes_retransmit=9 bytes_invalid=0 send_seq=167 receive_seq=166 retransmit_seq=2 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=20 stalled_bytes=0 freq=119990138 heater_bed: target=0 temp=0.0 pwm=0.000 print_time=2.560 buffer_time=0.212 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000
MCU 'mcu' is_shutdown: ?0
clocksync state: mcu_freq=120000000 last_clock=262206647 clock_est=(3520.790 215118672 119990138.906) min_half_rtt=0.000181 min_rtt_time=3521.128 time_avg=3520.789(0.018) clock_avg=215118672.948(2212577.267) pred_variance=10614411703.692
Dumping serial stats: bytes_write=3431 bytes_read=3566 bytes_retransmit=9 bytes_invalid=0 send_seq=166 receive_seq=166 retransmit_seq=2 srtt=0.002 rttvar=0.001 rto=0.025 ready_bytes=35 stalled_bytes=5
Dumping send queue 100 messages
```

I have a support ticket also opened with Klipper: https://github.com/KevinOConnor/klipper/issues/1841

Current we do not know what direction to go.

Kevin @ Klipper stated in his ticket:

> Unfortunately, you've hit the same issue described in #1673 - the micro-controller is responding with nonsensical messages (MCU 'mcu' is_shutdown: ?0) which indicates some sort of severe corruption of the micro-controller state. We've tested a number of the SAMD51 chips (I have the SAMD51G19 and SAMD51J19 chips and I know of several users using the SAMD51N19) - there must be something wrong with the SAMD51P20 support, but it is unclear what that is.

Can you review an provide some advise?

[klippy.log](https://github.com/adafruit/uf2-samdx1/files/3444562/klippy.log)

ladyada, 2019-07-30:
hiya this is the repo for the bootloader. it doesnt seem like there's any bootloader issue right? you are able to upload code with bossac/UF2?

jalanjarosz, 2019-07-30:
Generally speaking, probably not, unless there is something in the basic config or bossa command that should be different.

The bossac command did seem to execute correctly

```
c:\Program Files (x86)\BOSSA>bossac -u -p COM7 --offset=0x4000 -w .\Metro\klipper.bin -v -b -R
Unlock all regions
Write 19168 bytes to flash (38 pages)
[==============================] 100% (38/38 pages)
Done in 0.328 seconds
Verify 19168 bytes of flash
[==============================] 100% (38/38 pages)
Verify successful
Done in 0.252 seconds
Set boot flash true
```

Is there another project I can post this issue for further review/suggestions?

ladyada, 2019-07-30:
it sounds like you're having issue with Klipper firmware, so your issue should be on that repo, not here as you are not having a problem with this firmware :)

closed 2019-07-30 by ladyada

jalanjarosz, 2019-07-30:
Is there another Adafruit project that we can get some help with the Grand Central M4 board regarding the 

> MCU 'mcu' is_shutdown: ?0

 issue?

ladyada, 2019-07-30:
nope. we did not write a single line of that code, we have never run Klipper!