adafruit/uf2-samdx1 #96: Creating a custom board

gh185085, 2019-12-11:
Hi All,

I apologize if this isn't the correct forum, but I'm attempting to use the UF2 bootloader on a custom board.

A little background:
On PC1
I have a SAM D21 Xplained Pro Evaluation Kit that uses a SAMD21J18A. My development environment is Windows 10. I'm able to build one of the examples ($> make BOARD=sparkfun-samd21-dev) and load this onto the Xplained without issue through AtmelStudio. Moreover, I can drag-and-drop a LED toggle example (after modifying the start address to "-Wl,--section-start=.text=0x2000") that's been converted to UF2 and see the LED toggle as expected.

Problem:
On PC2 (custom board running Windows 10 with a SAMD21E17D connected)
Now, I'm trying to use a custom board that uses a SAMD21E17D. I copied the board\sparkfun-samd21-dev, renamed the directory, and modified the board_config.h (mostly to see a different LED blink during boot mode).

I also modified .\uf2-samdx1\scripts\samd21j18a.ld as the Makefile seems hardcoded to this file. The change I made here was to modify the ram LENGTH = 0x00004000 (to match the datasheet) and change the stack size from 0x2000 to 0x1000.

I'm able to build and load the custom board through AtmelStudio onto my custom target (PC2); this time using a Atmel-ICE debugger. I can see the LED fade in/out and the device enumerate as a MSD with the current.uf2, index.htm, and info_uf2.txt files while it's in boot mode. Now, when I try to drag-and-drop my new uf2, nothing happens. After a minute or so, the SAMD MCU appears to reset (no double-reset though, so it doesn't re-enter boot mode)

I'm a bit stuck on how to proceed debugging this issue. Any guidance is much appreciated.

Thanks,
Greg

dhalbert, 2019-12-11:
Try not changing the stack size: you may be overflowing the stack. `https://github.com/adafruit/uf2-samdx1/tree/master/boards/trinket_m0` and `gemma_m0` are SAMD21E18A, so use one of those board defs as a starting point.

gh185085, 2019-12-12:
Hi Dan,

Thanks for the quick feedback.

Unfortunately, I'm still running into the same issue. I.e. the MCU enumerates with the htm, uf2, and txt file, but dragging and dropping a new uf2 to upgrade appears to cause a timeout and reset.

The only difference between the SAMD21E/G/J is the pin count (32/48/64 pins) and the SAMD21*18/17/16/15 is flash memory (256/128/64/32 KB) and RAM (32/16/8/4 Kb). The best I can tell, I'm looking at half the memory from my eval kit (SAMD21J18A) to my custom board (ATSAMD21E17D), but from what I can tell, there should be sufficient space.

I can copy and paste generic text file without issue, but the uf2 upgrade is still failing.

Any additional thoughts/suggestions are greatly appreciated.

Thanks,
Greg

zjwhitehead, 2019-12-30:
I had a similar problem when the wrong `LED_TX_PIN` was specified. Try removing all the PIN definitions and see if that helps.

spacehuhn, 2020-04-09:
I'm facing the same problems using the ATSAMD21**E17A**.
I'm using `trinket_m0` as base and tried modifying `samd21j18a.ld` like @gh185085 described.
No luck on my end either :(

dhalbert, 2020-04-09:
I have fixed some issues with the SAMD21 code, especially the updater, which may have been present if you were working off of master. PR #113 is in progress. You could try using that as a base for now.

dhalbert, 2020-04-09:
@spacehuhn and @gh185085 Could you point me to your repos? Did you reduce both the `rom` and `ram` values in the `.ld` file by half?

spacehuhn, 2020-04-10:
Thanks for the quick response!

I have a dev board with a ATSAMD21E**18**A (256KB ROM, 32KB RAM) that works with the trinket m0 bootloader.
I'm trying to make these changes to compile a uf2 bootloader for the same board, but with a cheaper ATSAMD21E**17**A (128KB ROM, 16KB RAM).

I'm working on the [fuse-repair-fixes](https://github.com/dhalbert/uf2-samdx1/tree/fuse-repair-fixes) branch now.

I edited [`samd21j18a.ld`](https://github.com/dhalbert/uf2-samdx1/blob/fuse-repair-fixes/scripts/samd21j18a.ld#L50) to:  
```
  rom      (rx)  : ORIGIN = 0x00000000, LENGTH = 0x00002000
  ram      (rwx) : ORIGIN = 0x20000000, LENGTH = 0x00004000
```

And [`samd21j18a_self.ld`](https://github.com/dhalbert/uf2-samdx1/blob/fuse-repair-fixes/scripts/samd21j18a_self.ld#L50) to:
```
  rom      (rx)  : ORIGIN = 0x00002000, LENGTH = 0x00010000
  ram      (rwx) : ORIGIN = 0x20000000, LENGTH = 0x00004000
```
  
I'm building using [gcc-arm-2017-q4](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/7-2017-q4-major-1-1) on ubuntu 18.04.

Running `make BOARD=trinket_m0` works.
Flashing the resulting bootloader .bin using a J-Link over Atmel studio also works. The board is getting recognized and the TRINKETBOOT drive appears.

But that's it. It doesn't take a uf2 file and I can't be flashed using Arduino. 
This is the part where I run into the same issue as previously mentioned:

>Unfortunately, I'm still running into the same issue. I.e. the MCU enumerates with the htm, uf2, and txt file, but dragging and dropping a new uf2 to upgrade appears to cause a timeout and reset.

gmta10101, 2020-11-01:
@gh185085 Would you be able to share your Atmel studio files? I've been trying to get this repo to build in atmel studio for weeks and I'm having no luck. Can you walk me through it?

dhalbert, 2021-02-19:
Closing this for now. Happy to re-open if further discussion is needed.

closed 2021-02-19 by dhalbert

nijupp24, 2021-10-02:
HI All
i am working on a custom project which will use the SAMD21E17D controller am very poor in firmware  any one can support me with 17D custom Bootloader ?? with led on PA28