adafruit/uf2-samdx1 #98: Triggering Bootloader from User Code

zjwhitehead, 2019-12-30:
I'm looking for a way to boot the Feather M0 into the UF2 bootloader without pressing the reset button twice.
I know how to reset the device through the watchdog but that just creates a reboot loop without entering the bootloader. 

How can I trigger a reboot into bootloader either from 
a) setting a fuse and rebooting?
b) customize the bootloader to detect a button/pin being held on boot
c) different way other than using the reset button

ladyada, 2019-12-30:
you can set the 'magic memory location' and then jump to address 0 (or WDT reset or however)

https://github.com/adafruit/CircuitPlaygroundFirmata/blob/e13918e8ea65ceee263e72caa3748790a073a24b/CircuitPlaygroundFirmata_Express_CodeOrg/CircuitPlaygroundFirmata_Express_CodeOrg.ino#L1442

closed 2019-12-30 by ladyada

tannewt, 2020-01-02:
Here is where we do it in CircuitPython: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/microcontroller/__init__.c#L66