adafruit/uf2-samdx1 #99: Feather M0 gets bricked

blkist, 2020-01-11:
Hello, I followed this tutorial (https://learn.adafruit.com/installing-circuitpython-on-samd21-boards/installing-the-uf2-bootloader) on 3 M0 Feather boards and it bricked them all. I followed the directions perfectly every time and I used the Arduino INO file that is posed in this repository. I think there is an issue with the INO file. Also, in case it helps, the boards I used most recently were the Adalogger M0. Thanks for looking into this.

caternuson, 2020-01-13:
What was the exact name of the `.ino` file you used?

Pinging @dhalbert

blkist, 2020-01-13:
update-bootloader-feather_m0-v3.7.0.ino

JetForMe, 2020-03-12:
I, too, bricked a Feather M0 using the .ino file created by this repo. I thought it was because of the changes I introduced in `check_start_application()` but maybe it’s the .ino.

dhalbert, 2020-03-12:
Please do not use .ino files. I removed them from the release because they are broken in several ways, and put a warning in the release notes for 3.7.0. The M4 .ino files never worked, and the M0 ones have several flaws.

JetForMe, 2020-03-12:
Well, I only cloned this repo yesterday, and they were in master, and I don't see any obvious warnings.

dhalbert, 2020-03-12:
The warnings are just here: https://github.com/adafruit/uf2-samdx1/releases/tag/v3.7.0. I'm in the middle of doing a lot of debugging and improvement on this, and was either going to fix the .ino's or remove them. Sorry it's not clearer. Most of our users are just using the downloaded .ino's, so I removed them for safety.

JetForMe, 2020-03-12:
Fair enough. I should be getting an ATMEL-ICE tomorrow, hopefully it's easy enough to re-flash these from macOS. I really need to get that other bug fixed.

PR #111 (2020-04-02): fix SAMD51 spurious writes; allow gcc9 compilation; improve delay() calibration
Fixes #95.
Fixes #86.
Fixes #99.

- When starting the bootloader, for SAMD51, busy-wait until detected voltage is at least 2.7V, then wait 100msecs for the voltage to stabilize, then set auto-reset if voltage goes below 2.7V. Busy-waiting prevents multiple resets while the voltage is rising or glitching. As noted in #95, I tested the spurious writes fix for a number of hours.
Note that we can't enable `BOD33` in the user page fuses, because of an erratum that doing that can prevent attaching a debugging probe.
- Se [... cut]
closed 2020-04-02 by tannewt