adafruit/uf2-samdx1 #104: build failures

jerryneedell, 2020-02-23:
I cloned the repo and get the following error whenever try a build. Am I missing something?
I  am using  "gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major)"

``` 
arm-none-eabi-gcc -mthumb -mcpu=cortex-m4 -O2 -g -DSAMD51 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMD51J19A__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/feather_m4 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/feather_m4 -Ilib/samd51/include/ src/main.c -o build/feather_m4/main.o
In file included from lib/samd51/include/samd51j19a.h:546,
                 from lib/samd51/include/sam.h:38,
                 from ./inc/uf2.h:7,
                 from src/main.c:76:
lib/cmsis/CMSIS/Include/core_cmFunc.h: In function 'main':
lib/cmsis/CMSIS/Include/core_cm4.h:85:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
lib/cmsis/CMSIS/Include/core_cmFunc.h:456:3: note: in expansion of macro '__ASM'
  456 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
lib/cmsis/CMSIS/Include/core_cm4.h:85:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   85 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler          */
      |                            ^~~~~
lib/cmsis/CMSIS/Include/core_cmFunc.h:456:3: note: in expansion of macro '__ASM'
  456 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
cc1: all warnings being treated as errors
Makefile:176: recipe for target 'build/feather_m4/main.o' failed
make: *** [build/feather_m4/main.o] Error 1
```

dhalbert, 2020-02-23:
It's the `"sp"` on line 456, which no longer works with gcc9. The MicroPython people found this. It will still work with gcc7.

I'm planning to do a lot of work on the bootloader after 5.0.0 is released. Thanks for noticing.

NightSkySK, 2020-03-14:
The same issue on mine. Looking forward to this update as gcc-arm-none-eabi-9-2019-q4-major is required to build CircuitPython builds.

dhalbert, 2020-04-09:
Fixed by #111.

closed 2020-04-09 by dhalbert