adafruit/uf2-samdx1 #107: Lessons learned after Fuses have gone bad(=0xffffffff)

msperl, 2020-03-08:
A few comments after my expedition into feather bootloader debugging:

One of the observations that I have had after erasing and flashing the bootloader Of a feather-m0 with the jlink-flash Makefile target was that I somehow  managed to set both user fuses to 0xffffffff.

The effect was that the feather in boot loader mode was resetting every second or so because of the hard enabled watchdog fuse.
This was observable on Linux as disconnects on the usb bus and then a reconnect message in domestic.
Just looking at the led showed a bit of dimming which was aborted and turned of and then the thing repeated.

It took me quite some time to figure this out and then a windows machine with Atmel studio to configure the fuses correctly (not to mention: research into what are the right fuse values).

So here maybe a few things for improvement:
* add a watchdog timer reset in a few essential places of the boot loader so that this WD-reset does not happen and you do not brick the board when you have no jtag at hand...
* document the typical fuses to use for each platform (probably boards/<BOARD>/board.mk as well as a readme, wiki page or similar)
* document in https://learn.adafruit.com/how-to-program-samd-bootloaders/programming-the-bootloader-with-atmel-studio what may be the reason for the observed flashing and that fuses need to get corrected
* the above link is a bit dated and the studio does not show the hex numbers for the bootloader protection any longer - document the correct “textual version” as well as a link to the recommended fuses.
* provide a way to update the fuses with/or similar to the jlink-flash target (and possibly others as well)
* maybe set the fuses to defaults in the bootloader if they are set badly...

All of this would have helped me resolve the issue quicker...

dhalbert, 2020-03-08:
Thanks for your issue. There is already fuse reset code in `selfmain.c`, which is used by the bootloader updater, but it should probably be added to the regular bootloader `main.c` as well. We understand this pretty well, and I have a number of bootloader improvements and bug fixes in process. We;'ve had more trouble writing all 1's to the SAMD51, but on that chip the watchdog interval is long enough when set to all 1's that it's possible to run the updater. On the SAMD21 the interval is much shorter, but setting the flags to all 1's was much rarer, so that fuse reset code was not well tested.

msperl, 2020-03-09:
Still I would recommend that all this is documented so that it can get found quickly if this is ever seen by a different user...

dhalbert, 2020-03-09:
Yes, I agree, though it's hard to diagnose in some cases. I need to pinpoint the exact symptoms on different boards. That is all part of what I hope to fix and document while working on the bootloader in the ner future.

dhalbert, 2020-03-09:
In the SAMD51 case, I have some short Arduino programs to print out the fuse values. For SAMD21, I'm not sure if it would be possible to load these. Hence we need bootloader fixes.

msperl, 2020-03-09:
At least on the m0 feather the diagnostics of "dimming and then stop" as well as the observations on the USB bus that there are disconnects are an easy indication that something has gone bad with the fuses...

just in case setting some watchdog resets in the code should also help getting to that point...

finson, 2020-03-14:
I got excited by the Zephyr - Adafruit announcement and decided to explore a little bit. Using Mac OS and the various tutorials from Adafruit and Zephyr, I got all the way to "west flash" and then ran into a series of issues actually trying to burn code to my Feather M0 Basic Proto.  Long story short, I finally got Blinky running using bossac, however double click reset now drops into boot loader mode for about 500 ms, then Blinky runs again.  Not bricked, but not helpful, either.   I have a jlink and I expect I'll be able to restore a working boot loader once I get it configured, but I wanted to second the suggestion above that the documentation highlight the effects of corrupting the fuses and provide relatively clear help on how to fix the issue. (I'm not complaining, I recognize that I shot myself in the foot while blundering around!)

baebjr, 2020-03-19:
I second that! BTW awesome work, thank you very much!

dhalbert, 2020-04-09:
If the fuses get set to all 1's the watchdog ALWAYS-ON bit gets set on reset, and so we can't clear the watchdog timer programmatically. I am trying to squeeze in some fuse reset code that will detect the fuses being all 1's, and reset them to some reasonable values. But we are very close to being out of space on the SAMD21 bootloader.

PR #113 (2020-04-10): SAMD21 fixes
This does two things:
-- Fixes regression that arrived at some point in SAMD21 updater, which caused bricked boards. Fixes #112. Tagging @pelikhan.
-- Adds a mini fuse resetter to SAMD21 boards to fix fuses if first fuse word is set to all 1's, which causes continuous watchdog resets. Fixes #107.

I cleaned up the `selfmain.c` fuse-writing code a lot in the process of getting it to work again on SAMD21. Some trickery was removed. I would have liked to use this code for the SAMD21 fuse resetter mentioned above, but it was too big, so I used a stripped-down version. There are only 3 [... cut]
closed 2020-04-10 by dhalbert