adafruit/uf2-samdx1 #112: Drive dissapears on v3.8.0 - circuitplayground - m0

pelikhan, 2020-04-08:
I dropped the v3.8.0 bootloader on a CPX M0. The device enters bootloader mode (all leds green) but it fails to show the drive:
- drive shows up for a 1 second on windows, then dissapears
- no drive at all on a mac

dhalbert, 2020-04-08:
Thanks -- I had tested this but something's going wrong. I've made 3.8.0 be a pre-release while I investigate. It seems to work fine with SAMD51.

dhalbert, 2020-04-08:
The SAMD21 fuses are getting set to `0xffffffff`, which is enabling a very short WDT (watchdog timer).

To get the board up again, write these values to the fuses (from my .gdbinit).
```define fusefix-samd21
  # sets BOOTPROT to protect 8kB
  set *0x00804000 = 0xD8E0C7FA
  set *0x00804004 = 0xFFFFFC5D
end
```

dhalbert, 2020-04-08:
Fix doesn't work, checking further. I'll stop rushing.

pelikhan, 2020-04-08:
No worries!

PR #113 (2020-04-10): SAMD21 fixes
This does two things:
-- Fixes regression that arrived at some point in SAMD21 updater, which caused bricked boards. Fixes #112. Tagging @pelikhan.
-- Adds a mini fuse resetter to SAMD21 boards to fix fuses if first fuse word is set to all 1's, which causes continuous watchdog resets. Fixes #107.

I cleaned up the `selfmain.c` fuse-writing code a lot in the process of getting it to work again on SAMD21. Some trickery was removed. I would have liked to use this code for the SAMD21 fuse resetter mentioned above, but it was too big, so I used a stripped-down version. There are only 3 [... cut]
closed 2020-04-10 by dhalbert