adafruit/uf2-samdx1 #114: No UF2 v. 3.9.0 updater for PewPew M4

dobra-dobra, 2020-04-15:
File update-bootloader-pewpew_m4-v3.9.0.uf2 is missing in the recent release. There is a link to this file in https://circuitpython.org/board/pewpew_m4/

deshipu, 2020-04-15:
I need to update the bootloader's display configuration for it to work properly with the makerfab's displays (they have different offsets). I also need to fix the RGB/BGR in stage and board definition, so that you get correct colors when using ``displayio``.

deshipu, 2020-04-17:
@dhalbert done at https://github.com/adafruit/uf2-samdx1/pull/115

dhalbert, 2020-04-17:
Thanks, I'll merge this and make a new release soon (v3.9.1), but I want to add some kind of "notes" field on the circuitpython.org pages to tell people who have v3.9.0 on other boards not to bother to upload to v3.9.1, since the only difference is a board addition.

deshipu, 2020-05-17:
@dhalbert Any problems with this? Can I help somehow?

dhalbert, 2020-05-17:
Fixed by #115.

closed 2020-05-17 by dhalbert