adafruit/uf2-samdx1 #124: [BUG] First user code memory reset

GMagician, 2020-06-18:
I think version 3.9.0 has not solved the user program area memory corruption.
It's 2nd time I see my board that starts into boot mode. First time I just reloaded my program, 2nd time I checked memory with sagger, here is what I got from damaged state:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
D9 DA 02 00 D9 DA 02 00  D9 DA 02 00 00 00 00 00

and this on working state:
00 00 04 20 ED DA 02 00  D9 DA 02 00 D9 DA 02 00
D9 DA 02 00 D9 DA 02 00  D9 DA 02 00 00 00 00 00

dhalbert, 2020-06-18:
I tested various SAMD51 boards overnight, toggling power on and off, and did not see this. What are your fuse settings? And how are you powering the board (regular USB?)?

GMagician, 2020-06-18:
> What are your fuse settings?

Need to check

> And how are you powering the board (regular USB?)?

That's a good question. I'm using Marlin firmware than I have an external 24V PSU and a shield that gives 8V to AGCM4, but I also have, sometime, usb cable connected

dhalbert, 2020-06-18:
The new code will not exit the bootloader until the chip voltage is at least 2.7V for 100ms. But after that it doesn't check. If it's unstable for longer than that, the old problem could still occur. We have not seen this problem with regular USB or regular battery power.

GMagician, 2020-06-18:
Here are my fuses at 0x804000:
39 92 9A F6 
80 FF EC AE  
FF FF FF FF 
10 40 80 00
10 40 80 00
FF FF FF FF

GMagician, 2020-06-18:
I thought that due to your considerations. If you think PSU has no stable tension there is nothing that can be done.

GMagician, 2020-06-27:
Ok, did it again...system was working. Powerd off and on re-switching on ... code lost.
A question, during your test did you discovered who writes the first user memory?

dhalbert, 2020-06-27:
The flash write is due to inadequate power; it's a hardware problem. It's not due to any particular code that's actually trying to do a write, but it does seem to be dependent on the code that's executing.. See
https://community.atmel.com/forum/samd51-intermittent-spurious-writes-zeros-nvm-power-cycle
https://community.atmel.com/forum/samd51-not-starting-after-power-cycle
and I also had a dialogue with MicroChip about this, where they said it was due to missing filter capacitors on VDD. They said:

> The one time we came across a similar issue of flash areas written with 0x00 upon frequent power cycles, it was an issue at the power supply stage.
> 
> This was with a different M0+ device, where the recommended filter capacitor was not installed on the Vddcore.
> 
> 
> Your observation that it takes a fast power toggle of multiple times to reproduce the issue, suggests that this could be a similar case.
> 
> For the device to function properly, a stable core voltage is required.
> 
> Unstable core voltage or fluctuations would cause such unpredictable behaviour.

I tested this extensively with a Metro M4 and Feather M4 and conventional USB power, power cycling the board until test by using the EN pin (which disables the 3.3V regulator). I had another microcontroller doing the power-cycling, and monitoring whether the user program on the tested board was changing the output on a particular pin. If the user program got smashed, then the pin change would not happen, and I knew flash had been corrupted.

I was able to reproduce the original problem with this test setup, and then after adding the code to wait for 100ms of stable power, the same test setup ran overnight with no problems.

1. Just to double-check, have you confirmed by looking at INFO_UF2.TXT that your bootloader is indeed 3.9.0?
2. Are you able to put a storage scope or Saleae on the power to see if there are glitches when you switch it on?

GMagician, 2020-06-28:
Here is info_uf2.txt content:
_UF2 Bootloader v3.10.0 SFHWRO
Model: Grand Central M4 Express
Board-ID: SAMD51P20A-GrandCentral-v0_

I'll try to see what happens on power. My problems are present when powered by an external 8V power, no usb. Maybe voltage raise is different on 3.3V when such power supply is used vs usb one. This issue has occured other 4 times after firtst post. I suspect it happens when I turn off, not on because last time I re-powered it with usb first and I got system into boot mode.

If my doubt is confirmed it may be some power glitch when button is opening contacts...I'll further investigate on that.

I'll see what I can do to monitor power supply

dhalbert, 2020-06-28:
>I suspect it happens when I turn off, not on because last time I re-powered it with usb first and I got system into boot mode.

What firmware are you running? Do you have a link to it? Does it change the BOD33 level, or does it disable BOD33? Or does it do other things to the power controller peripheral? That could cause a problem.

Before the bootloader goes to the user program, it sets BOD33 to 2.7V, and enables reset on brownout. See lines 231-234:

https://github.com/adafruit/uf2-samdx1/blob/e585be56f10481a37ef060036adaa3fb1047f9f4/src/main.c#L200-L234

If the power supply is unstable for more than 100ms, you could try lengthening the 100ms wait time on line 226.

You could also try adding more capacitance across the input jack, the 5V line, or the 3.3V line. The input and the 5V line already have 47uF each, the 3.3v line has 10uF (plus a bunch of decoupling caps). And it could be flaky switching, too, as you suggest.

https://learn.adafruit.com/adafruit-grand-central/downloads#schematics-3015168-2

GMagician, 2020-06-28:
I'm using Marlin and AFAIK it doesn't change BOD registers.

Wouldn't be better intead of:
```c++
 if (RSTC->RCAUSE.bit.POR || RSTC->RCAUSE.bit.BODVDD) { 
     do { 
         // Check again in 100ms. 
         delay(100); 
     } while (SUPC->STATUS.bit.BOD33DET); 
 }
```

something like:
```c++
 if (RSTC->RCAUSE.bit.POR || RSTC->RCAUSE.bit.BODVDD) { 
     int cnt = 0;
     do { 
         // Wait power is stable at least for 100ms. 
         delay(1); 
         if (SUPC->STATUS.bit.BOD33DET)
           cnt = 0;
         else
            ++cnt;
     } while (cnt < 100); 
 }
```

in this case if during power up tension is not stable it way for a 100ms stability (not sure how BOD33DET works but if it reports current state it may works)

dhalbert, 2020-06-28:
I think `STATUS.bit.BOD33DET` is the output of the comparator. It's not latching. It could be checked continuously while keeping time. Or the code could just turn on reset-on-brownout before waiting 100ms, so that going below the BOD33 level would cause the whole thing to start over again. My thinking in the code above was that 100ms was basically a debounce time.

You could try one of the above and see if it improves you situation. But I do think it's worth seeing what's going on with your power supply or switching.

GMagician, 2020-06-28:
> My thinking in the code above was that 100ms was basically a debounce time

In present code you are only sure that after first time you reach power level after 100 ms is still in range, but nothing assure you it's stable.

> But I do think it's worth seeing what's going on with your power supply or switching.

I'll try to check also this

dhalbert, 2020-06-28:
> > My thinking in the code above was that 100ms was basically a debounce time
> 
> In present code you are only sure that after first time you reach power level after 100 ms is still in range, but nothing assure you it's stable.

Right, my assumption was that 100ms is enough time to achieve stability. I was looking at the rise time when powered by USB, and was considering various battery-powered scenarios.

dhalbert, 2020-06-28:
From `MarlinCore.cpp`:
```c
  // Check startup - does nothing if bootloader sets MCUSR to 0
  byte mcu = HAL_get_reset_source();
  if (mcu &  1) SERIAL_ECHOLNPGM(STR_POWERUP);
  if (mcu &  2) SERIAL_ECHOLNPGM(STR_EXTERNAL_RESET);
  if (mcu &  4) SERIAL_ECHOLNPGM(STR_BROWNOUT_RESET);
  if (mcu &  8) SERIAL_ECHOLNPGM(STR_WATCHDOG_RESET);
  if (mcu & 32) SERIAL_ECHOLNPGM(STR_SOFTWARE_RESET);
  HAL_clear_reset_source();
```
If you never see the "Brownout" message, sounds like it might be on power-down, as you suspect.

GMagician, 2020-06-28:
oh yes...you are right, I forgot it, think I wrote it....my memory.... I didn't noticed such message, but I'm honest, I never searched for it on the boot up messages. I'll take care to check

GMagician, 2020-07-01:
I'm testing my proposed solution to see it something changes. Already reset 2 or 3 times and no BOD messages ever seen

GMagician, 2020-07-17:
With my proposed fix it seems better but not solved.
I changed boot code to signal, by the neopixel, the BOD status and I can't see color when switched on, I'm pretty sure it happens when turning off (i see the neopixel red color). I suspect a not clean open of power switch that bring system to a BOD.

closed 2020-09-07 by GMagician