adafruit/uf2-samdx1 #132: Support for SAMD21E16A

ruffner, 2020-08-10:
I was attempting to compile the ```trinket_m0``` variant of this bootloader for a custom board using an E16A variant and I get the following errors after changing the chipname in the ```board.mk``` file. Presumably this is due to the lessened flash/ram compared to the E18A (64k/8k vs 256k/32k). Seeing as the bootloader is only 8k though I thought it should work.
```
Building trinket_m0
echo "src/flash_samd21.c"
src/flash_samd21.c
arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunctio
n-sections -D__SAMD21E16A__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wcha
r-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -W$
loat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls
 -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -
I./inc/preprocessor -I./boards/trinket_m0 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/trinket_m0 -Ilib/samd21/samd21a/include/ src/flash_samd21.c -o build/trinket_m
0/flash_samd21.o
In file included from src/flash_samd21.c:1:0:
./inc/uf2.h:301:73: error: division by zero [-Werror=div-by-zero]
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                                                         ^
./inc/uf2.h:306:1: note: in expansion of macro 'STATIC_ASSERT'
 STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
 ^~~~~~~~~~~~~
./inc/uf2.h:301:42: error: enumerator value for '_static_assert_306' is not an integer constant
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                          ^
./inc/uf2.h:299:24: note: in definition of macro 'CONCAT_1'
 #define CONCAT_1(a, b) a##b
                        ^
./inc/uf2.h:301:33: note: in expansion of macro 'CONCAT_0'
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                 ^~~~~~~~
./inc/uf2.h:306:1: note: in expansion of macro 'STATIC_ASSERT'
 STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
 ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:177: build/trinket_m0/flash_samd21.o] Error 1
```

I see that the cause is ```FLASH_NUM_ROWS * 4``` is apparently not equal to ```FLASH_NB_OF_PAGES``` in the E16A variant. Is this a necessary condition for UF2 ? If you comment line 306 in ```inc/uf2.h``` it compiles but obviously not an ideal solution.

ruffner, 2020-08-11:
After reading the comments on issue #96 I am able to get my computer to recognize the board once the bootloader is flashed, but am experiencing the same problem of not being able to upload to it from Arduino. 

I saw that changing the stack size was discouraged, however the total RAM is only 8k is and the default stack is 8k. when compiling UF2 this causes an overflow. This being said, is it possible to run UF2 on a micro with only 8k RAM?

dhalbert, 2020-08-11:
It was not a goal to get UF2 to run on such small processors, so we didn't try it. But certainly try lowering the stack size incrementally to see if you can get it to work, and make sure it makes sense in the context of the the `.ld` file.

You could also add a little code to fill the stack with a sentinel value, run the bootloader on a '18 processor, set a breakpoint before it resets to the user program, and then see the maximum stack excursion. There is some code like this in https://github.com/adafruit/circuitpython, see around these lines:
```c
py/stackctrl.h
50:const char MP_MAX_STACK_USAGE_SENTINEL_BYTE;

py/stackctrl.c
66:const char MP_MAX_STACK_USAGE_SENTINEL_BYTE = 0xEE;
95:	*p-- = MP_MAX_STACK_USAGE_SENTINEL_BYTE;

shared-module/ustack/__init__.c
40:    while (*p++ == MP_MAX_STACK_USAGE_SENTINEL_BYTE) { }
```

ruffner, 2020-08-11:
Excellent, thank you for the insight, I will keep working on it. 

Separately, would you be able to explain the difference between the ```*.ld``` and ```*_self.ld``` linker scripts?

dhalbert, 2020-08-11:
`selfmain.c` _et al_ are for the update-bootloader*.uf2 builds, which are executables that run in the program space to update the bootloader. So you load one of these .uf2's on a board with an existing bootloader, and it rewrites the bootloader region with the new version. It has to unset the BOOTPROT fuses to do this, and when it's done it sets them to re-protect the region.

ruffner, 2020-08-11:
Okay that makes sense. So since I am testing functionality by targeting the device from the Arduino IDE, only changes that are mode to 'non-self' files will affect the resulting functionality.

dhalbert, 2020-08-11:
That is right, but if you don't fix the `self` files too, then you won't be able to update the bootloader conveniently in the future.

HpLightcorner, 2020-09-20:
@ruffner did you find out how much stack is needed?
At least I have made some experience with BOSSA... Which indeed needs either customization of the Bootloader or BOSSA itself as far as I understood.

ruffner, 2020-09-21:
@HpLightcorner I did not. It was simpler for my application to just swap chips. I did try lowering the stack size incrementally until the bootloader would compile, however user code would still not load. Dan offered some excellent advice about measuring the actual stack used but I got the other chips sooner and didn't experiment with that approach. The '17D might have enough ram to support it, but I think the '16D is just to puny,

HpLightcorner, 2020-09-22:
@ruffner I will further investigate the necessary resources for the 17D - currently I was focusing on the SAM-BA part and actually I think changing Arduino:XYZ to Arduino:XZ and reducing Stack to 4 kB could allow to run the boot loader on the 16D as well (at least for the SAM-BA part as no additional free RAM is needed for buffering). Would be interesting if these changes work on the 16D, maybe you can give it a try? (as I have no 16D board laying around...) The upload time on the 16D increases, but that is something a 16D user could live with. For UF2, the question is if additional free RAM Is needed? 

But yeah the 16D is for sure a challenge to run on...
Reducing the stack to 4kB actually worked for me, both BOSSA and UF2 are running now without any issues.

ruffner, 2020-09-24:
I actually replaced all the 16D chips with 18 on the four board I brought up...sorry. Glad to hear you are making progress!

dhalbert, 2021-02-19:
I am going to close this for now, but feel free to ask for it to be re-opened if someone figures out how to make it work on low-RAM processors.

closed 2021-02-19 by dhalbert