adafruit/uf2-samdx1 #136: USB VID/PID request

theacodes, 2020-08-24:
Hi folks,
I'm building a new open source Eurorack module around the SAMD21 and while it doesn't make sense for it to run CircuitPython, I would like to use the uf2 bootloader to allow easily updating the firmware. It would be lovely if I could have PIDs for the bootloader and firmware. If not, that's okay - I have some other options, I just wanted to ask here first because it would be lovely for all of the Winterbloom modules to have the same VIDs and the others have the Adafruit one. :)

Thanks!

DefProc, 2020-08-25:
If it's open source, have you looked at: [pid.codes](http://pid.codes/)?

dhalbert, 2020-08-25:
@theacodes We are happy to give you one. I'm just behind on my issues :).

theacodes, 2020-08-25:
Thanks, Dan. No rush. 🙂

dhalbert, 2020-08-25:
```
VID 0x239A

PID 0x00C3  # bootloader
PID 0x80C3  # arduino
PID 0x80C4  # circuitpython  [can be unused, no problem; there is just a pattern we follow]
```

closed 2020-08-25 by dhalbert

theacodes, 2020-08-26:
Thank you so much!