adafruit/uf2-samdx1 #142: Adafruit Grand Central M4 (SAMD51P20A) bootloader issues and USB capture logs

greymfm, 2020-09-21:
Hi,
We have purchased **300 pieces Adafruit Grand Central M4 boards** for our Ardumower project. We have bootloader issues with **all of these boards** using **many different computers** (all all different combinations of OS's, Windows, Linux etc.). We have tried out all available Grand Central M4 bootloader versions (v3.30, v3.7.0,  v3.9.0, v3.10.0 ) . Same issues. 

Problem: The UF2 bootloader is 90% of the time not showing up any USB devices when loading via double-pressing RESET on the Adafruit Grand Central M4 board. The Neopixel LED stays red and the small LED is quickly pulsating. We have flashed a user program using Arduino IDE and the Arduino blink example code but it's difficult to flash when the bootloader does not show any USB devices 90% of the time.

We have captured several USB log dumps attached using 'usbmon' to be viewed with 'wireshark'.
[usblog.tar.gz](https://github.com/adafruit/uf2-samdx1/files/5254793/usblog.tar.gz)

What else could we do to help finding the cause for such a problem? :-) 

Thanks, Alexander

dhalbert, 2020-09-21:
An early batch of Grand Central M4's had the main timing crystal installed upside down, which caused USB timing issues. See the first picutre below.. The component near the "E" in "POWER" should not have two pads visible: it should look like the second picture. This caused problems similar to what you describe. If you search in https:forums.adafruit.com for << grand central crystal >> you will find several threads with more pictures. The other possibility is bad USB cables, but I would assume you have tried alternate cables. If not, please do. Please open a thread on the forum for this issue.
[image]
[image]

dhalbert, 2020-09-21:
Also, could you say from where you purchased these boards?

greymfm, 2020-09-21:
Thank you! The crystals are soldered correctly (not upside down). Also, yes, we tried different USB cables. We purchased the boards via Mouser Electronics. I will post a thread on the forum. Thank you for your time!

greymfm, 2020-09-21:
Here's the Adafruit forum thread link for this post: https://forums.adafruit.com/viewtopic.php?f=63&t=169701

greymfm, 2020-09-21:
Features say: 'In-memory logging for debugging - use the logs target to extract the logs using openocd' -
I can connect to the board using OpenOCD  -  Any examples on how could I extract logs using OpenOCD ? :-)

ladyada, 2020-09-23:
@dhalbert please check the thread when you're back!

dhalbert, 2020-09-29:
Replied in thread: https://forums.adafruit.com/viewtopic.php?f=63&t=169701

dhalbert, 2020-10-01:
Closing for now due to further research by OP that only a few boards seem to have problems. Please reopen as necessary. 

https://forums.adafruit.com/viewtopic.php?f=63&t=169701&p=831529#p831525

closed 2020-10-01 by dhalbert