adafruit/uf2-samdx1 #157: MatrixPortal M4 download/build missing

StarWitch, 2021-01-14:
I notice from the MatrixPortal M4 documentation that there are instructions to update the ESP32 module, but I'm not noticing a download for the SAMD51 itself.

https://circuitpython.org/board/matrixportal_m4/

https://learn.adafruit.com/adafruit-matrixportal-m4/updating-esp32-firmware

ladyada, 2021-01-14:
this?
https://learn.adafruit.com/upgrading-esp32-firmware/upgrade-an-airlift-all-in-one-board
[image]

StarWitch, 2021-01-14:
Perhaps I'm misunderstanding the purpose of the SAMD51 in the MatrixPortal M4, but I assumed that both the SAMD51 and the ESP32 had separate bootloaders.

The Passthru bootloader on that page did let me update the ESP32 module through `esptool.py`, but the bootloader for the SAMD51 itself seems to be missing from this repo's release page. Currently, my MatrixPortal M4's SAMD51 is on this version:

```
UF2 Bootloader v1.23.1-adafruit.1-328-gf06693a SFHWRO
Model: Matrix Portal M4
Board-ID: SAMD51J19A-MatrixPortal-v0
```

While the ESP32 module is on [this release](https://github.com/adafruit/nina-fw/releases/tag/1.7.1).

ladyada, 2021-01-14:
ah! ok yes its is!
@dhalbert plz add to release artifacts!

StarWitch, 2021-01-21:
For what it's worth, I built a DIY firmware for a second MatrixPortal M4 I bought. Here's the configuration that worked for me: 

`board_config.h`
```
#ifndef BOARD_CONFIG_H
#define BOARD_CONFIG_H

#define VENDOR_NAME "Adafruit Industries"
#define PRODUCT_NAME "MatrixPortal M4"
#define VOLUME_LABEL "MATRIXBOOT"
#define INDEX_URL "http://adafru.it/4745"
#define BOARD_ID "SAMD51J19A-MatrixPortal-v0"

#define CRYSTALLESS    1

#define USB_VID 0x239A
#define USB_PID 0x0035

#define LED_PIN PIN_PA14

#define MTX_ADDRA PIN_PB07
#define MTX_ADDRB PIN_PB08
#define MTX_ADDRC PIN_PB09
#define MTX_ADDRD PIN_PB15
#define MTX_ADDRE PIN_PB13

#define MTX_CLK PIN_PB06
#define MTX_LAT PIN_PB14
#define MTX_OE PIN_PB12

#define MTX_R1 PIN_PB00
#define MTX_G1 PIN_PB01
#define MTX_B1 PIN_PB02
#define MTX_R2 PIN_PB03
#define MTX_G2 PIN_PB04
#define MTX_B2 PIN_PB05

#define BOARD_NEOPIXEL_PIN PIN_PA23
#define BOARD_NEOPIXEL_COUNT 1

#define BOOT_USART_MODULE                 SERCOM5
#define BOOT_USART_MASK                   APBAMASK
#define BOOT_USART_BUS_CLOCK_INDEX        MCLK_APBDMASK_SERCOM5
#define BOOT_USART_PAD_SETTINGS           UART_RX_PAD1_TX_PAD0
#define BOOT_USART_PAD3                   PINMUX_UNUSED
#define BOOT_USART_PAD2                   PINMUX_UNUSED
#define BOOT_USART_PAD1                   PINMUX_PB30D_SERCOM5_PAD1
#define BOOT_USART_PAD0                   PINMUX_PB31D_SERCOM5_PAD0
#define BOOT_GCLK_ID_CORE                 SERCOM0_GCLK_ID_CORE
#define BOOT_GCLK_ID_SLOW                 SERCOM0_GCLK_ID_SLOW

#endif
```

`board.mk`
```
CHIP_FAMILY = samd51
CHIP_VARIANT = SAMD51J19A
```

Copied it from `pyportal_m4`, added the missing definitions, and then changed `SAMD51J20A` to `SAMD51J19A` and the BOOT_USART_PAD0/1 pins to line up with I2C (since that seems to be what they represented on the PyPortal). Works fine with CircuitPython 6.1.0-rc1 as far as I can tell, using the Moon Clock example.

dhalbert, 2021-02-19:
@ladyada I booted a MatrixPortal and see that its UF2 drive is `MATRIXPORTAL`. Did you make a board def for it, or did you just hack an existing one? I looked in https://github.com/ladyada/uf2-samd21 and don't see one in `master` or in a branch. Tnx.

ladyada, 2021-02-21:
PR made

dhalbert, 2021-02-21:
Fixed by #163

closed 2021-02-21 by dhalbert