adafruit/uf2-samdx1 #161: NVMCTRL not declared in this scope

VaFlyer, 2021-02-19:
Trying to load bootloader via Arduino sketch to a Feather M0 with LoRa, using "update-bootloader-feather_m0-v3.12.0."  I'm following "Using LoraWAN and The Things Network with CircuitPython" instructions. [https://learn.adafruit.com/using-lorawan-and-the-things-network-with-circuitpython/circuitpython-installation]
I keep getting "'NVMCTRL' not declared in this scope." and the upload process exits.  Thinking I'm missing some step or file here.  I tried commenting out the offending line but then another NVMCTL line is declared not in scope.
What am I doing wrong?

dhalbert, 2021-02-19:
Don't comment anything out. That kind of error is usually because:
- You need to set the Board to "Feather M0"; it is still some other non-SAMD board
-  You need to install the SAMD Board Support Packages. Install the Arduino SAMD package first, then the Adafruit one:
  - "Arduino SAMD Boards"
  - "Adafruit SAMD Boards"

Also I'd recommend installing the latest Arduino IDE before doing the above as well.

dhalbert, 2021-02-21:
Let us know if this did not work.

closed 2021-02-21 by dhalbert

VaFlyer, 2021-02-21:
Your advice was exactly right. Thanks for helping the newbie.

Virginia Stouffer
Transformational Technologies