adafruit/uf2-samdx1 #165: Print statements only to serial?

mirandazellnik, 2021-03-13:
Hello, wondering if anyone can help me figure this out. I'm using the Adafruit Pygamer device, with the latest bootloader and CircuitPython stable releases. I'm making some graphical programs, but I'm running into a serious issue.

Whenever I print anything, the PyGamer seems to automatically display it over whatever was currently on the screen. This results in constant flashing when running a graphical program that was intended to only communicate via text **over the serial connection**.

I can't figure out a way to directly access the serial connection from the CircuitPython code, however - I can only access it by printing, which then writes over whatever was displayed over the screen. Is there a way to directly deal with the serial connection and communicate that way?

If I don't find a way, I'll settle for logging to a file - except I'm having issues with that too. I'll open an issue to do with that shortly. Thanks for any help you can give!

ladyada, 2021-03-13:
hi, github is not where we do tech support and project help. please post to the forums. thanx!

closed 2021-03-13 by ladyada

tannewt, 2021-03-16:
FYI, forum is here: https://forums.adafruit.com/viewforum.php?f=60