adafruit/uf2-samdx1 #174: Usage on a board with no reset button

izzy84075, 2021-06-16:
I have a handful of battery-powered custom boards, and my intention is to not have a reset button. I'm hoping to be able to distribute UF2 files instead of the project source, just to make sure everything ends up with the same code.

I have code to trigger the bootloader after holding down the button I do have for ~30 seconds. However, if you /accidentally/ trigger this, there seems to be no way to exit the bootloader without disconnecting the battery or writing a program to the board.

Is there some way to trigger a reset/application start after, say, 30 seconds of no USB connection, or by plugging it into a computer and then disconnecting?

dhalbert, 2021-06-16:
You could modify the existing bootloader to keep a timer running and reboot to the application program if there was no activity after a while. https://github.com/adafruit/Adafruit_nRF52_Bootloader did this, though it was removed inhttps://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/166. The nRF bootloader also start the application program if USB has not ben enumerated after 3 seconds: https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/6126d1192ef2fde7bff2eaa2df68c22639230cb7/src/main.c#L266

There is no timeout built in to this SAMD bootloader, but similar timeouts could be added.

izzy84075, 2021-06-18:
Digging around some, I think the "reset if no USB enumeration" functionality _almost_ exists already.

In the case of the USE_SINGLE_RESET configuration, resetHorizon gets set in [check_start_application()](https://github.com/adafruit/uf2-samdx1/blob/master/src/main.c#L113) , and then timerTick gets [called](https://github.com/adafruit/uf2-samdx1/blob/master/src/utils.c#L49) while waiting for [USB_Ok](https://github.com/adafruit/uf2-samdx1/blob/master/src/cdc_enumerate.c#L550) to enumerate, and if timerHigh passes resetHorizon, it resets into the app.

If it does enumerate, it gets cleaned up [here](https://github.com/adafruit/uf2-samdx1/blob/master/src/main.c#L280) to prevent it resetting until programming is complete.

I'm fairly certain this is also the reset method used by MSC programming, as seen [here](https://github.com/adafruit/uf2-samdx1/blob/master/src/fat.c#L254).

So I think I can do this with ~2 lines of code, just to set resetHorizon to a default value on startup, and then to ensure it gets cleaned up regardless of USE_SINGLE_RESET after enumeration.

I believe this is a very different implementation from how the nrf52 bootloader had it, as it looks like that one could timeout even after enumeration and reset, causing host computer errors. This is only intended to reset in the case of there never being a host computer connected.

Would this be something possibly useful to add a configuration option for in a PR? It can easily default to the existing behaviour of not initializing resetHorizon/initializing it to 0 to prevent it automatically resetting out of the bootloader, and only add the timeout behaviour when explicitly configured to do so in the board configuration.

nunomiguelferreira, 2022-03-11:
If it is still possible I would like really this behaviour to be enabled, since at the moment it is possible for the boards to be stuck in bootloader mode forever.

A timeout of 60 seconds or so, would vastly increase reliability of the bootloader.

dhalbert, 2022-03-11:
We have no plans to add this, but would be happy to entertain a pull request as an optional feature enabled by a compile-time flag.

nunomiguelferreira, 2022-03-11:
@izzy84075 Don't know if you made any progress with this, but would love to have the possibility to do so!

Also a separate branch with the compile time flag would be nice, if its easy to do (at the momentm wouldn't know how to do it).

xyzzy42, 2022-09-15:
I did this for our device.  I used the watchdog timer and fed it based on USB or UART progress.  This way it still resets if USB is connected and then stops working.