adafruit/uf2-samdx1 #175: SAMD21E17A support

mgiacomelli, 2021-07-11:
Lately the SAMD21G series processors have become scarce (one supplier quoted me a 12 month lead time) due to the chip shortage.  To address this I was hoping to get an Arduino environment up on some of the easier to find SAMD variants like the SAMD21E17A which is the same expect with half the RAM/flash.  I added a new Trinket M0 based board with a new board_config.h, board.mk with the correct and edit samd21j18a.ld to have half of the RAM/flash.  Finally I added a ATSAMD21E17A device to Atmel.SAMD21_DFP.pdsc, although I'm not sure if that was needed. 

I can compile and flash the bootloader with ST-LINK, and it will appear to work and present the UF2 MSC volume.  I can even copy an updated .uf2 image to that volume and then boot into the updated image.  However, as soon as I unplug the device and then plug it back in, the board goes dead until I reflash it with the ST-LINK.  Apparently something about a cold boot crashes it where a hot boot does not.

I read through #132 and #96 which describe similar problems on the smaller SAMD boards, but these eventually get to BOSSA, which I believe I am not even using yet since I'm just flashing with the ST-LINK.  At this point I'm a little stuck.  Is there some way I can trouble shoot what happens on power up?

davidvondle, 2022-05-20:
Did you get anywhere with this?  I have some ATSAMD21E17A boards on the way and am hoping I can get a bootloader on them.

ademuri, 2023-03-26:
I have a [fork of this repo](https://github.com/candykingdom/uf2-samdx1) which adds support for some custom boards based on the ATSAMD21E17A, if you want an example of how to make this work.