adafruit/uf2-samdx1 #176: Can not Compile "SAMD21G17A" with  "#define CRYSTALLESS  1"

ZanderFoster, 2021-08-20:
Hello, I am trying to compile the boot loader for my "SAMD21G17A" board. However when I change #define BOARD_ID "SAMD21G18A"   to the SAMD21G17A chip it refuses to compile with the CRYSTALLESS option. 
the SAMD21G18A seems to compile without problem which is odd considering the only differences between the two chips is the flash size... 

Here is what happens when I run the make command 
`
$ make BOARD=scb
Building scb
echo "src/flash_samd21.c"
src/flash_samd21.c
arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMD21G17A__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/scb -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/scb -Ilib/samd21/samd21a/include/ src/flash_samd21.c -o build/scb/flash_samd21.o
In file included from src/flash_samd21.c:1:
./inc/uf2.h:338:73: error: division by zero [-Werror=div-by-zero]
  338 | #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
      |                                                                         ^
./inc/uf2.h:343:1: note: in expansion of macro 'STATIC_ASSERT'
  343 | STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
      | ^~~~~~~~~~~~~
./inc/uf2.h:338:42: error: enumerator value for '_static_assert_343' is not an integer constant
  338 | #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
      |                                          ^~~~~~~~~~~~~~~
./inc/uf2.h:336:24: note: in definition of macro 'CONCAT_1'
  336 | #define CONCAT_1(a, b) a##b
      |                        ^
./inc/uf2.h:338:33: note: in expansion of macro 'CONCAT_0'
  338 | #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
      |                                 ^~~~~~~~
./inc/uf2.h:343:1: note: in expansion of macro 'STATIC_ASSERT'
  343 | STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
      | ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:181: build/scb/flash_samd21.o] Error 1
`

ZanderFoster, 2021-08-20:
After digging into the code and looking at other boards I tried adding 
```
#define FLASH_NUM_ROWS 512
```
to my board_config.h and it seemed to compile correctly.

dhalbert, 2021-08-20:
Glad you got it working. See some of the closed issues for further issues that might crop up with SAMx17.

closed 2021-08-20 by dhalbert