adafruit/uf2-samdx1 #177: Unlock bootloader for j-link/SWD debugging

zunkworks, 2021-09-08:
Hello, I'm trying to do some debugging using a j-link on a Feather M4, but am running into a locked bootloader. How can I install the bootloader such that it is unlocked?

I'm following [this Adafruit tutorial](https://learn.adafruit.com/make-a-simple-debugging-featherwing-for-the-m0?view=all), which is for the Feather M0, but should be close enough. When I try to connect to the device with the j-link using JLinkGDBServer it returns an error:

>Connecting to target...
>ERROR: Could not connect to target.
>Target connection failed. GDBServer will be closed...Restoring target state and closing J-Link connection...
>Shutting down...
>Could not connect to target.
>Please check power, connection and settings.

If I connect more directly with JLinkExe, I get this popup message:
>J-Link V7.52d Info
>Secured Atmel SAMD device detected.
>For debugger connection the device needs to be unsecured.
>Note: Unsecuring will trigger a mass erase of the internal flash.
>Do you want to unsecure the device?

If I try unsecuring the device I get the same `Mass erase done!` message over and over until I get the previous popup message. 

According to [this forum post](https://forums.adafruit.com/viewtopic.php?f=57&t=142170&p=707151#p707151) it is possible to have an unlocked bootloader, and the locking/unlocking can be controlled by the UF2 file. 

I am able to compile the bootloader, I'm just not sure what I need to change to prevent it from locking after I load it on the device.

dhalbert, 2021-09-08:
It is not the bootloader that is locked: it is that the NVMCTRL "security bit" appears to be set. However, this is not true: the J-Link is having trouble talking to the chip, and the flash memory appears to locked.

I saw this myself a couple of weeks ago when attempting to use the J-Link with a SAMD51, and was surprised, since it worked well in the past. It might be a J-Link software problem due to a recent update. I will try this again and see if it happens again.

(The bootloader is protected against overwriting (BOOTPROT), but this is not a problem for J-Link access: we debug chips all the time with BOOTPROT enabled.)

zunkworks, 2021-09-09:
Ok, thank you for checking/helping.

zunkworks, 2021-09-28:
Were you ever able to check to see if you experience the same problem? Just to confirm, what are the minimum needed connections to the J-Link?

dhalbert, 2021-09-28:
Sorry, I was having trouble initially, and meant to revisit this, and then forgot.

I just tried connecting to a Metro M4 (SAMD51J19) and it's working fine now. Just today I installed the latest version of the J-Link software, version V7.54d. When I connected the J-Link, it asked to update the firmware on the J-Link, which I did, and it appears to be working fine.

You can download the latest software for J-Link from: https://www.segger.com/downloads/jlink/#J-LinkSoftwareAndDocumentationPack

dhalbert, 2023-02-03:
closing this as apparently fixed

closed 2023-02-03 by dhalbert