adafruit/uf2-samdx1 #184: Compiling for the SAMDA1G16B

jlbrian, 2022-01-29:
This is more of a question than an issue.  I am trying to compile this for the SAMDA1G16B and it appears that everything builds without any issues, however, the bootloader does not work.  The first thing I am wondering is if there is a way to debug the code as it runs so that I can step through it line by line.  The second question is if there is someone that has had some experience adding support for other microcontrollers has any advice.  Below is a link to my project including the build folder and modified source, the compiler output, and screenshots of the fuses and registers after transferring the .bin to the target.

I have run examples in Atmel Studio written for the SAMDA1 Xplained Pro board and am able to connect with the computer and interact with the microcontroller through a CDC example, so I don't believe that it's a hardware problem.  

Any help would be appreciated.

https://github.com/jlbrian/uf2-samdx1-1

> Building my_board_a1_g3
> echo "#define UF2_VERSION_BASE \"v3.14.0-dirty\""> build/my_board_a1_g3/uf2_version.h
> echo "src/flash_samda1.c"
> src/flash_samda1.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/flash_samda1.c -o build/my_board_a1_g3/flash_samda1.o
> echo "src/init_samda1.c"
> src/init_samda1.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/init_samda1.c -o build/my_board_a1_g3/init_samda1.o
> echo "src/startup_samda1.c"
> src/startup_samda1.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/startup_samda1.c -o build/my_board_a1_g3/startup_samda1.o
> echo "src/usart_sam_ba.c"
> src/usart_sam_ba.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/usart_sam_ba.c -o build/my_board_a1_g3/usart_sam_ba.o
> echo "src/screen.c"
> src/screen.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/screen.c -o build/my_board_a1_g3/screen.o
> echo "src/images.c"
> src/images.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/my_board_a1_g3 -Ilib/samda1/samda1b/include/ src/images.c -o build/my_board_a1_g3/images.o
> echo "src/utils.c"
> src/utils.c
> arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMDA1 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMDA1G16B__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/my_board_a1_g3 -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ib [... cut ...]
> arm-none-eabi-objcopy -O binary build/my_board_a1_g3/update-bootloader-my_board_a1_g3-v3.14.0-dirty.elf build/my_board_a1_g3/update-bootloader-my_board_a1_g3-v3.14.0-dirty.bin
> python3 lib/uf2/utils/uf2conv.py -b 8192 -c -o build/my_board_a1_g3/update-bootloader-my_board_a1_g3-v3.14.0-dirty.uf2 build/my_board_a1_g3/update-bootloader-my_board_a1_g3-v3.14.0-dirty.bin
> 10051 39.26171875 39
> Converting to uf2, output size: 19968, start address: 0x2000
> Wrote 19968 bytes to build/my_board_a1_g3/update-bootloader-my_board_a1_g3-v3.14.0-dirty.uf2.

[image]
[image]

[image]
[image]

dhalbert, 2022-01-29:
The SAMDA1G16B only has 8kB of RAM. The default stack size is 8kB. People have had trouble building for other smaller chips: see for instance #132.

I also don't know what's different about the peripherals and other aspects of the A1 vs 21: there may be other changes you need to make based on internal differences.

jlbrian, 2022-01-29:
From everything that I have seen, they are almost identical.  The files in the pack for the samda1 seem to be identical, and they samd21 and the two microcontrollers have a common datasheet.  

https://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf

dhalbert, 2022-01-30:
Then it is probably the RAM size issue.

jlbrian, 2022-01-30:
I'll take a look at it some more.  The microcontrollers are definitely similar.  I can build this project targeting the SAMD21G16B and then flash the SAMDA1G16B with the .elf file and the application seems to run fine.   I get a communication and mass storage device.

https://github.com/watterott/SAM-BAR

jlbrian, 2022-01-30:
Adjusting the stack size fixed the problem, below is the current stack setting,  Now I have to see if there is enough space to do anything useful with it.

dhalbert, Thanks for your response.

[image]

closed 2022-01-30 by ladyada