adafruit/uf2-samdx1 #187: circuit playground express 3.14 macos Monterey

PhusionDev, 2022-03-26:
When i downloaded and flashed [update-bootloader-circuitplay_m0-v3.14.0.uf2]
on Mac Monterey 12.2.1 to my circuit playground express ATSAMD21 @ 48MHz
the firmware updates and shows the new 3.14 firmware but stays stuck in bootloader mode -- code.org maker app can detect the board but not connect to it. I had to download the firmware from the windows instructions [FIRMATA_2018-2-19.UF2] and flash that to get it working in the application.

PhusionDev, 2022-03-26:
I was using information from the adafruit website about updating the bootloader on macos and it wasn't very clear that I shouldn't be using this firmware if I was not using makecode. The way it was presented I thought that I needed this file to use code.org Maker App.

closed 2022-03-26 by PhusionDev

dhalbert, 2022-03-26:
Could you point to the confusing section? The bootloader update is only for the UF2 bootloader: it is not the software for MakeCode or any of the other application programs.

PhusionDev, 2022-03-26:
When I updated the bootloader using this firmware, all of the lights on my CPX remained green (bootloader mode) even when i disconnected the device. I don't know if this is intended behavior or not, but after flashing this I couldn't use it with Code.org Maker App

There is a section of the adafruit site for the CPX which says "Update the Bootloader (Mac Users Especially)" that seems as though it's something I should be doing.

You can see said section here: https://learn.adafruit.com/adafruit-circuit-playground-express/updating-the-bootloader on the navigation menu on the left.

dhalbert, 2022-03-26:
When you update the bootloader, it erases anything else on the board, and leaves only the bootloader.  So it would have erased FIRMATA or whatever else was there. Then you can reload FIRMATA, CircuitPython, a MakeCode program, or an Arduino program. I will try to make this clearer in the instructions. Thanks for pointing this out.