adafruit/uf2-samdx1 #192: Makefile issues

davepruitt, 2022-08-11:
I just successfully built the grand central M4 bootloader, but it took some tweaking of the makefile. It's possible that the issues are related to me using Windows rather than Linux/Unix, but I figured I would report the issue anyway just in case.

There are a few issues that I encountered.

First: the makefile includes the following lines which are supposed to create the uf2_version.h file:

```
$(BUILD_PATH)/uf2_version.h: Makefile
	echo "#define UF2_VERSION_BASE \"$(UF2_VERSION_BASE)\""> $@
```

In order for these lines to work, I first had to manually create the **./build/grandcentral_m4/** folder. Otherwise it would crash. I expected it to be able to create these folders on its own. I am confident I was working in a folder with write permissions.

Additionally, the uf2_version.h file that was created did not have the expected text. The actual contents of the created file were the following:

```
"#define UF2_VERSION_BASE \"v3.14.0-5-g779ccde-dirty\""
```

And I presume that the intention was to create a file with the following contents instead:

```
#define UF2_VERSION_BASE "v3.14.0-5-g779ccde-dirty"
```

I had to manually change the file contents after the file was created to get it to work.

Next, the makefile requires **python3** on lines 177 and 187. This is not listed in the build requirements in the read-me for this repository, and it should be.

Additionally, I actually _do_ have Python 3 installed (version 3.9.11 to be exact, installed with conda), and it is the _only_ Python installation I have on my machine, _and_ it is in the path, however the Python binaries are not called python3. Rather, the Python binary is simply called python. Therefore, the call to execute python3 failed because there is no binary called python3.

It may be wise to adapt the makefile so that it could use a binary called python3 or a binary called python, depending on what is installed on the user's system. For me, as a workaround, I just edited the makefile on my machine to call python instead of python3.

dhalbert, 2022-08-11:
The `build` folder issue sounds like a Windows idiosyncrasy, though I don't know why that would happen. It works fine on Linux.

Python can be called `python` or `python3` depending on whether you install it from the Windows store or from python.org. I can't remember which does which.

davepruitt, 2022-08-11:
Right, hence I was able to edit the makefile fairly easily to get it to successfully build. I just wanted to point it out since some users may try to build with a binary called "python", which would cause the makefile to fail.

Anyways, at the very elast it may be wise to add Python 3 as a build requirement in the read-me for this repository, because it is not currently listed.

adityathakekar, 2022-12-04:
@davepruitt thanks for documenting can you also share how you edited the make file for python? 
I am also facing the same issues.

adityathakekar, 2022-12-04:
Thanks, the issue is solved. 
1. Check if there are two versions of python running.
2. [Python check for environment path related issues](https://stackoverflow.com/questions/65348890/python-was-not-found-run-without-arguments-to-install-from-the-microsoft-store)
3. In the readme need to add the python requirement.