adafruit/uf2-samdx1 #193: macOS 13.0 (Ventura) Finder gives error when uploading UF2 files

dhalbert, 2022-10-31:
The new macOS release 13.0 (Ventura), has trouble uploading files to “fake” USB drives that are used for firmware updates on microcontroller boards. These fake USB drives are presented by UF2 bootloaders and also ARMmbed DAPlink bootloaders. UF2 bootloaders take .uf2 files and are used on Adafruit boards, Raspberry Pi Pico boards, and many other manufacturer’s boards. ARMmbed DAPlink takes .hex files and is used on the micro:bit and other boards.

The problem is in the macOS Finder. Dragging a firmware file to the drive produces “error code 100093”. This error apparently occurs because the Finder is trying to copy extended attributes to the drive.

You can work around the problem by copying the file using the command line in Terminal. Use the cp -X command. For example, to upload a Circuitpython UF2 file to a Raspberry Pi Pico, you would do:

```sh
% cp -X adafruit-circuitpython-raspberry_pi_pico-en_US-7.3.3.uf2 /Volumes/RPI-RP2
```

For further information and similar workarounds, see:

https://blog.adafruit.com/2022/10/31/uploading-uf2-files-with-macos-13-0-ventura-apple-microbit_edu-raspberry_pi-circuitpython/
https://github.com/raspberrypi/pico-sdk/issues/1081
https://github.com/orgs/micropython/discussions/9794
https://support.microbit.org/support/solutions/articles/19000139995-finder-error-100093-when-using-drag-and-drop-to-transfer-a-micro-bit-hex-file
https://github.com/ARMmbed/DAPLink/issues/982

This problem has been reported to Apple. If you’d like to report it, you can use the macOS Feedback Assistant tool.

This is a tracking issue - we don't expect to fix it ourselves.

aallan, 2022-11-01:
We've also now written this up as a blog post, https://www.raspberrypi.com/news/the-ventura-problem/.

lurch, 2022-12-15:
This has now been fixed (for UF2 files at least) in the latest Ventura update - see https://www.raspberrypi.com/news/the-ventura-problem/

aallan, 2022-12-15:
Drag-and-drop of UF2 files for Raspberry Pi Pico and other RP2040-based boards has been fixed as of macOS Ventura 13.1. The fix does not resolve the issue where .hex files >1MB fail to write as file blocks are still out of order.

dhalbert, 2023-02-03:
Still fixed in Ventura 13.2. Closing for now.

closed 2023-02-03 by dhalbert