adafruit/uf2-samdx1 #194: Linking issue

davepruitt, 2022-11-08:
I am running Ubuntu on Windows Subsystem for Linux. When I try to build the `zero` bootloader, it successfully compiles all the files, but fails at the linking step.

Here is the error I am receiving:
```
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: section .binfo LMA [0000000000001ff0,0000000000001fff] overlaps section .relocate LMA [0000000000001ef8,00000000000020c7]
collect2: error: ld returned 1 exit status
make: *** [Makefile:161: build/zero/bootloader-zero-v3.14.0-5-g779ccde-dirty.bin] Error 1
```

Any suggestions on how to fix this?

Note: I am actually able to successfully build this on _Windows_, but when I try building on _Ubuntu_ it gives me this linking error.

dhalbert, 2022-11-08:
We normally build with arm-none-eabi-gcc 10.2.1, the same toolchain we use for CircuitPython. See https://learn.adafruit.com/building-circuitpython for how to install the toolchain. Could you try with that?

r-corm, 2024-10-03:
I have the same issue. I'm using  arm-none-eabi-gcc 10.2.1 and Ubuntu 22.04

dhalbert, 2024-10-03:
I just tried `make BOARD=zero` (after `make clean`) with the `arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi` toolchain, and it worked for me on Ubuntu 24.04. Are you changing any of the settings before compiling?

r-corm, 2024-10-03:
> I just tried `make BOARD=zero` (after `make clean`) with the `arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi` toolchain, and it worked for me on Ubuntu 24.04. Are you changing any of the settings before compiling?

Thanks for the quick reply. I will try again with those versions to see if that helps. 

No, I'm not making any changes before compiling, is there something I should be changing?

I am using WSL like the OP, so it could have something to do with that.

dhalbert, 2024-10-03:
> is there something I should be changing?

Nope, not at all. I just wanted to make sure I was compiling what you were compiling.