adafruit/uf2-samdx1 #197: Reading logs - how????

davepruitt, 2023-02-10:
I am having some issues with the bootloader and it would be helpful to get logging working. Unfortunately, there is not really any explanation or tutorial on how to read the logs that the bootloader generates. 

In `uf2.h`, I have set `USE_LOGS` to be equal to 1:
```
#define USE_LOGS 1
```

I have noticed from reading `utils.c` and `uf2.h` that the logs seem to just be stored in a 4096 byte memory buffer. From `uf2.h`:
```
struct LogStore {
    int ptr;
    char buffer[4096];
};
extern struct LogStore logStoreUF2;
```

From `utils.c`:
```
struct LogStore logStoreUF2;
```

Is there an easy way to read these logs from my computer? If the response is "use OpenOCD", that doesn't really give me any more information. Are there any tutorials or documentation anywhere about how to read those logs?

My setup:
- Windows 10 machine
- Atmel ICE debugger
- My custom board is very similar to the Adafruit Grand Central M4, but I'm using a SAMD51P19A.

The specific issues I am having with the bootloader (the reasons why I want to get logging working):
- For my custom board, I've built a bootloader (which seems to work) and I've also created an Arduino board definition. I am writing code and uploading "sketches" to my board using the Arduino IDE. Upon the first upload (when the board is in the "bootloader" state with no sketch yet loaded on), I am successful at uploading the board. Then, if I try to edit the sketch and re-upload it to the board through the Arduino IDE, it fails. Typically, the Arduino IDE should send a 1200 bps signal to the board, which the board should recognize and reset itself into bootloader mode in preparation for a new sketch. It seems that when it resets into bootloader mode, the USB port disappears from my computer and doesn't show back up. If I cycle power to the board, then it will enter the sketch again (and the sketh's USB COM port will show up). The only way I can get back into the bootloader is to use Microchip Studio to completely flash the SAMD chip with a new bootloader and erase the sketch.

dhalbert, 2023-02-11:
The logging mechanism was developed by the original authors of this code at Microsoft. I too have made some attempt to use the logging, but didn't quite succeed. In the `scripts/` directory there is some Javascript code that may be informative. If that doesn't help, you could inquire in an issue in the original repo: https://github.com/microsoft/uf2-samdx1

mfvmx, 2024-03-22:
I was able to get it to work by installing node.js and slightly changing the dbdtool.js to point to my arduino openocd install directory by making this change:
```
    // let openocdPath = pkgDir + "tools/openocd/0.10.0-arduino1-static/"
    let openocdPath = pkgDir + "tools/openocd/0.11.0-arduino2/"
```
Then updating the file search in dbdtool.js to this:
```
        for (let ln of mapFile.split(/\r?\n/)) {
            // let m = /^\s*0x00000([0-9a-f]+)\s+(\S+)/.exec(ln)
            let m = /^\s*0x0*([0-9a-f]+)\s+(\S+)/.exec(ln);
            if (m && m[2] == logSym) {
```
Then running this in the command prompt:

`node dbgtool.js firmware.map`

dhalbert, 2024-07-21:
Closing as resolved. Thanks for the followup.

closed 2024-07-21 by dhalbert