adafruit/uf2-samdx1 #207: VS code ESP IDF components issue

Jimis333, 2023-09-13:
I'm trying to take an example code from ESP IDF (https_advanced_ota) and put it inside a folder called "ota" so that it's not in the root and keep it tidy. 
This creates a problem with the includes and declarations. I updated the cmakelists.txt files accordingly which is a bit of a nightmare. I think I managed to do most correctly, but I still get an error of a file not included, although I explicitly tell the compiler where the file is by providing the path in the #include statement and the c_cpp_properties.json 

Can someone take a look and help me find the problem. I'm obviously missing something here.

[image]
[folders.zip](https://github.com/adafruit/uf2-samdx1/files/12594193/folders.zip)

dhalbert, 2023-09-13:
Hi - did you mean to file this issue in this repo? This is the repo for the SAMD UF2 bootloader, which isn't related to ESP-IDF for Espressif at all.

Jimis333, 2023-09-13:
ups sorry I didn't notice that. I will try to move it

closed 2023-09-13 by Jimis333